stm32-rs/stm32f3xx-hal #1: Integrate to stm32-rs org

TeXitoi, 2019-04-04:
Hi!

I've found your crate while looking around (checking if https://olkb.com/parts/qmk-proton-c can be supported by rust). Great work!

I think this crate can go to the stm32-rs organisation. You'll join the org, and can have feedbacks of the other members on your common hal.

cc @adamgreig @therealprof

therealprof, 2019-04-04:
Ooh, nice. Indeed, @dfrankland, if you'd like to join our org you'd be very welcome to do so, no strings attached.

dfrankland, 2019-04-04:
@TeXitoi, thanks for asking about the Proton C! I've actually got Rust running on it already 😄

https://github.com/dfrankland/proton-c

@therealprof I am surprised y'all found me so fast! I was hoping to get this into the org after I got everything working 👍 I don't mind moving it over now either.

dfrankland, 2019-04-04:
Btw, I think that some of the GPIO pins need to be double-checked. I also have not tested all of the physical devices.

therealprof, 2019-04-04:
@dfrankland Whenever you're ready.

dfrankland, 2019-04-07:
We're moved over now 🎉

closed 2019-04-12 by dfrankland