stm32-rs/stm32f3xx-hal #10: Support for HSE oscillator?

lylecheatham, 2019-06-30:
Hi there,

I was wondering if support for the HSE oscillator in RCC for STM32F3 (F303 in my particular case) was planned. Technically, the LSE oscillator isn't fit to support USB, so an external oscillator is required for usb.

I'm hoping to create an application that uses USB so this would be a blocking item.

My current idea for a work around is to configure in STMCube and then just link in and call the clock config functions.

Let me know if this isn't the right place for a question like this.

dfrankland, 2019-06-30:
Hello! Thanks for asking. I would personally love to support USB as best as possible, so it would be great to support the HSE oscillator. Currently there's some work being done for USB support in https://github.com/stm32-rs/stm32-usbd.

Here's an example using https://github.com/stm32-rs/stm32-usbd for STM32F303:
https://github.com/stm32-rs/stm32-usbd-examples/blob/master/example-stm32f303vc/src/main.rs

lylecheatham, 2019-06-30:
I was chatting with some people on the rust-embedded IRC, and we had the idea of making it easier to take values from the STM32CubeMX solver (all the PLLMUL, PLLSRC, PPRE1, etc.) and put that into the RCC.

Essentially an adjacent method with less handholding, but still a step up from bit twiddling on pointers, which is what is currently necessary to use USB and/or the HSE oscillator. Does that sound reasonable?

dfrankland, 2019-06-30:
I don't know much about it, but I am open to anything :+1:

dfrankland, 2019-07-01:
For anyone else interested, I found a nice, recent, related blog article about integrating code generated by STM32 Cube MX into Rust:
https://nercury.github.io/rust/embedded/experiments/2019/01/27/rust-embedded-02-measuring-the-clock.html

dfrankland, 2019-07-21:
Looks like #18 will add some initial support for the HSE oscillator.