stm32-rs/stm32f3xx-hal #9: SPI Clock/Signal not right

ibutra, 2019-06-30:
Hi,

I switched from alt-stm32f30x-hal to this hal because the code looked easier for me to contribute to. But after switching I have a problem which I am sure what it is cause by, probalby my stupidity.

I am using the same SPI Code as with the other hal except of course the necessary adaptions in parameters for the functions etc. but I can't get a valid SPI Signal out of my STM anymore.
Instead it looks like this:
[image]

I am obviously missing 3 clock cycles for a single byte.
I was tryting to figure out what is different but I can't see any meaningful differences between this library and the alt library in terms of initialization, so it must be something in my code I guess, so here is my initialization of the SPI, can you spot something wront?
```

    let stm = hal::stm32::Peripherals::take().unwrap();
    let core = cortex_m::Peripherals::take().unwrap();
    //This is done for the peripherals to play nice I think!
    let mut flash = stm.FLASH.constrain();
    let mut rcc = stm.RCC.constrain();

    //clock configuration
    let clocks = rcc.cfgr
        .sysclk(64.mhz())
        .pclk1(32.mhz())
        .pclk2(32.mhz())
        .freeze(&mut flash.acr);

    let mut delay_timer = Delay::new(core.SYST, clocks);

    //spi
    let mut gpioa = stm.GPIOA.split(&mut rcc.ahb);
    let mosi = gpioa.pa7.into_af5(&mut gpioa.moder, &mut gpioa.afrl);
    let miso = gpioa.pa6.into_af5(&mut gpioa.moder, &mut gpioa.afrl);
    let sck = gpioa.pa5.into_af5(&mut gpioa.moder, &mut gpioa.afrl);
    let cs = gpioa.pa4.into_push_pull_output(&mut gpioa.moder, &mut gpioa.otyper);
    let mut spi = hal::spi::Spi::spi1(stm.SPI1, (sck, miso, mosi), MODE_0, 1.mhz(), clocks, &mut rcc.apb2);
    spi.send(0b101010);
```

dfrankland, 2019-06-30:
Which MCU are you using? Some of the code is conditional based on which MCU you use, which may be causing this issue.

ibutra, 2019-06-30:
I am using the STM32F303K8

Sh3Rm4n, 2019-12-27:
This was maybe fixed via https://github.com/stm32-rs/stm32f3xx-hal/pull/35.

Another guess is that the Clock configuration could be wrong? I'll have to
check, when I have my discovery board available again.

Sh3Rm4n, 2020-03-07:
So I've tried to reproduce this on the stm32f3-discovery board and it seems to work as expected. 

[image]
(MISO and MOSI are connected in this example, so MISO mirrors MOSI) 

I will close this issue for now. However, feel free to reopen it, if something still doesn't work on your configuration.

closed 2020-03-07 by Sh3Rm4n