stm32-rs/stm32f3xx-hal #22: Unable to initialize RCC

waicool20, 2019-11-02:
Currently I am unable to initialize RCC on my MCU because of some problems with the HAL. The MCU in question is STM32F303CCT6 and it has a 8 MHz HSE attached, I am initializing it with 

```rust
    let clocks = rcc.cfgr.use_hse(8.mhz()).sysclk(72.mhz())
```

The problem code in the hal library I've identified is this snippet from `rcc.rs`
```rust
        let pllsrcclk = self.hse.unwrap_or(HSI / 2);

        let pllmul = (2 * self.sysclk.unwrap_or(pllsrcclk)) / pllsrcclk;
        let pllmul = cmp::min(cmp::max(pllmul, 2), 16);
        let pllmul_bits = if pllmul == 2 {
            None
        } else {
            Some(pllmul as u8 - 2)
        };

        let sysclk = pllmul * self.hse.unwrap_or(HSI) / 2;
```
My current fix is to remove the 2* and /2 from line 2 and the last line by modifying `rcc.rs`

```rust
        let pllsrcclk = self.hse.unwrap_or(HSI / 2);

        let pllmul =  self.sysclk.unwrap_or(pllsrcclk) / pllsrcclk;
        let pllmul = cmp::min(cmp::max(pllmul, 2), 16);
        let pllmul_bits = if pllmul == 2 {
            None
        } else {
            Some(pllmul as u8 - 2)
        };

        let sysclk = pllmul * self.hse.unwrap_or(HSI);
```

I'd like to ask why the multiplication and division were there in the first place, if they aren't needed I can put up a pull request to include the fix. 

The problem is that `pllmul` becomes 18 which ends up getting coerced to 16, so the value of `sysclk` and `hclk` ends up being 64MHz which is obviously not what we want. This is not the problem in itself as the MCU in question is fully capable of running those peripherals at that speed but down the line `pclk1` ends up becoming 64MHz as well. Which then fails the assert
```rust
        let ppre1 = 1 << (ppre1_bits - 0b011);
        let pclk1 = hclk / u32(ppre1);

        assert!(pclk1 <= 36_000_000);
```
Which is also expected behaviour since pclk1 does not support more than 36MHz so in the end the problem lies upstream.

russellmcc, 2019-12-01:
I believe I've fixed these issues in the following PR: https://github.com/stm32-rs/stm32f3xx-hal/pull/31

Sh3Rm4n, 2020-03-07:
Hey @waicool20 could you try, if it works with the current version, if you have the time :)

Sh3Rm4n, 2020-05-06:
I'll close this for now. Feel free to comment, if the issue still exists.

closed 2020-05-06 by Sh3Rm4n