stm32-rs/stm32f3xx-hal #36: Justify `.write` use

strom-und-spiele, 2019-12-03:
After following a few discussions on issues, it occurred to me that using `.write` reliably produced unwanted behavior. 
The only way I know how to solve this is to grep for `.write` and justify it's uses by adding comments or correct the wrong usages. This is some what ugly and requires future commits to hold to this practice, but it'd emphasize the importance of paying attention here.

I'd be happy to hear a cleaner solution and am wiling to implement it ( in the long run though - It seems like my life wants to be busy right now ;) )
I guess it won't be too much work at the current state of the repo (`$ rg "\.write" src/* | wc -l   → 26`)

IamfromSpace, 2019-12-03:
I totally agree with this.  When new to the write/modify I was using write inappropriately and it caused me endless headaches.

It seems like we would want modify in basically every case anyway.  I believe that the only reasons to use write are implicit reset and optimization.  Probably we’d want to prefer explicitness and optimization should be done with care—so they should be justified when done.

On a broader scope, based on the API, I originally used “write” as if it were modify when learning.  Maybe that’s just me being dumb, but it seems I’m not the only one.  Maybe it should be named “writeOrReset” or something to clarify what it does and steer users to modify (deprecating write).  I understand that that goes far beyond this package :)

Sh3Rm4n, 2019-12-03:
Totally agree. I was confused by the `write` method as well. This ultimatly comes down to how svd to rust works and there seems to be some ideas, how to tackle this. https://github.com/rust-embedded/svd2rust/pull/258#issuecomment-494698256

`write_or_reset` and names alike, which avoid confusion, should be the way to go, to replace `write`. Maybe we should propose a change to [svd2rust](https://github.com/rust-embedded/svd2rust)?

strom-und-spiele, 2019-12-06:
Did it on my fork: https://github.com/strom-und-spiele/stm32f3xx-hal will push it as a PR here if given access

dfrankland, 2019-12-06:
Hmm? No access necessary to make PRs 👍 go ahead and make one

strom-und-spiele, 2019-12-06:
hmm I got
```
]$ git push -u origin/justify_write
fatal: 'origin/justify_write' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```
with my git config containing
```
[remote "origin"]
        url = https://strom-und-spiele@github.com/stm32-rs/stm32f3xx-hal.git
        fetch = +refs/heads/*:refs/remotes/origin/*
```

sorry if i'm missing the obvious.

dfrankland, 2019-12-06:
I think you need to double check your remotes using

```bash
git remote -v
```

If `origin` is set correctly try to do this:

```bash
git push --set-upstream origin justify_write
```

dfrankland, 2019-12-06:
OH! @strom-und-spiele I see the issue now. You didn't actually fork this repo, you just copied it. Could you fork it using the "Fork" button at the top of this repo's page and then push your changes to that fork? Finally, you'll be able to open a pull request.

strom-und-spiele, 2019-12-08:
Thanks, that clarified a lot for me :D

closed 2020-03-18 by Sh3Rm4n