stm32-rs/stm32f3xx-hal #40: HardFault with semihosting hprintln

lucazulian, 2019-12-21:
I was trying to print some text out but with the code below I have a `Breakpoint 2, 0x08002690 in HardFault_ ()` error
```
#![no_main]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rt as rt;
extern crate stm32f3xx_hal as hal;
use core::panic::PanicInfo;
use cortex_m_semihosting::hprintln;
use hal::stm32;

#[entry]
fn main() -> ! {
    hprintln!("Hello, world!").unwrap();
    loop {}
}
```

I'm using a `STM32F303K8` board with these dependencies:
```
[dependencies]
cortex-m = "0.6.1"
cortex-m-rtfm = "0.4.3"
panic-halt = "0.2.0"
alloc-cortex-m = "0.3.5"
embedded-hal = "0.2.3"
generic-array = "0.13.2"
nb = "0.1.0"
cortex-m-semihosting = "0.3.5"

[dependencies.cortex-m-rt]
version = "0.6.11"
features = ["device"]

[dependencies.stm32f3xx-hal]
version = "0.3.0"
features = ["stm32f303", "device-selected"]
```

Sh3Rm4n, 2019-12-22:
Which version of rustc did you use? Do you use nightly?

`alloc-cortex-m = "0.3.5"` does not seem to be buildable without nightly.

Even without it, I can't compile your example.

```console
error: cannot find attribute `entry` in this scope
  --> src/main.rs:11:3
   |
11 | #[entry]
   |   ^^^^^

warning: unused import: `core::panic::PanicInfo`
 --> src/main.rs:7:5
  |
7 | use core::panic::PanicInfo;
  |     ^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: unused import: `hal::stm32`
 --> src/main.rs:9:5
  |
9 | use hal::stm32;
  |     ^^^^^^^^^^

error: `#[panic_handler]` function required, but not found

error: aborting due to 2 previous errors

error: could not compile `issue-40`.

To learn more, run the command again with --verbose.
```

This is because neither `panic_halt` nor `cortex_m::entry` is used.

```patch
6a7,9
> extern crate panic_halt;
>
> use rt::entry;
```

Did you try the [cortex_m_quickstart](https://github.com/rust-embedded/cortex-m-quickstart) [hello](https://github.com/rust-embedded/cortex-m-quickstart/blob/master/examples/hello.rs) example?

Actually, I couldn't create a runnable version of your provided example. Strangely the binary did not contain debug symbols, but that's another issue I did not debug.

I could get the quickstart example working.

Adjusting the target in `.cargo/config` and changing `memory.x` flash size:

```linker
  FLASH : ORIGIN = 0x08000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 40K
```

Nonetheless, this issue doesn't have anything to do with this crate, as it is not using the hal at all and it only depends on the cortex_m runtime.

Sh3Rm4n, 2020-03-07:
I'll close this issue for now, as it is not really relevant for this project.
However, don't hesitate to ask if you still have issues.

closed 2020-03-07 by Sh3Rm4n

DarianHarrison, 2023-03-16:
// full description of error: http://nnarain.github.io/2021/08/02/Setting-up-per-package-targets-in-Rust.html

```rust
#![no_main]
#![no_std]

// had to add this to prevent linker errors
use stm32f3xx_hal as _;

use panic_halt as _;

use cortex_m_rt::entry;
use cortex_m_semihosting::hprintln;

#[entry]
fn main() -> ! {
    hprintln!("Hello, world!");

    loop {}
}
```