stm32-rs/stm32f3xx-hal #46: docs.rs build not working

dfrankland, 2019-12-29:
Related: https://github.com/stm32-rs/stm32f1xx-hal/issues/154

Reported by @strom-und-spiele: https://github.com/stm32-rs/stm32f3xx-hal/pull/45#issuecomment-569492210

Likely caused by the addition of this:
https://github.com/stm32-rs/stm32f3xx-hal/blob/7e0d7e67617fdb12174f76872a00bd2f54a6eb01/.cargo/config#L8

dfrankland, 2019-12-29:
We can change the `default-target` to `thumbv7m-none-eabi`, but docs.rs does not support it yet.

```diff
  [package.metadata.docs.rs]
  features = ["stm32f103", "rt", "stm32-usbd"]
+ default-target = "thumbv7m-none-eabi"
```

Until that's added, does anyone see issues with making `default-target = "x86_64-unknown-linux-gnu"`?

dfrankland, 2019-12-29:
Related docs.rs issues:
https://github.com/rust-lang/docs.rs/issues/343

strom-und-spiele, 2019-12-29:
As we currently don't have a `[package.metadata.docs.rs]` section, I don't see any harm in adding it. I'd suggest building for the discovery board as we provide all examples for it.

Does this one sum it up?
```toml
[package.metadata.docs.rs]
features = ["stm32f3DISCOVERY", "rt"]
# thumbv7m-none-eabi is not a supported target on docs.rs,
# however the target is not relevant for docs.
default-target = "x86_64-unknown-linux-gnu"
```

dfrankland, 2019-12-29:
We do have a section here:
https://github.com/stm32-rs/stm32f3xx-hal/blob/7e0d7e67617fdb12174f76872a00bd2f54a6eb01/Cargo.toml#L13

I think the comment you wrote also makes sense to add.

strom-und-spiele, 2019-12-29:
Ah sorry, was falsely checking the `.cargo/config` file.
I added the  `stm32f3DISCOVERY` feature a while ago when working with the adc Module (as I found it more convenient; https://github.com/stm32-rs/stm32f3xx-hal/pull/47/files#diff-80398c5faae3c069e4e6aa2ed11b28c0R52 )  Sorry for the confusion.

Disasm, 2020-01-06:
Seems to be a bug in docs.rs, because it tries building on the default target, but does not fetch a rustup target for it. Adding `default-target = "x86_64-unknown-linux-gnu"` should help.

rubberduck203, 2020-01-11:
I was able to confirm that the approach of the default target to the package meta-data works.

```toml
[package.metadata.docs.rs]
default-target = "x86_64-unknown-linux-gnu"
```

https://docs.rs/crate/stm32f3-discovery/0.1.3/source/Cargo.toml

closed 2020-03-07 by Sh3Rm4n