stm32-rs/stm32f3xx-hal #48: Point external documentation to this crate

dfrankland, 2020-01-01:
We should likely update [`awesome-embedded-rust`][] and also [`stm32f30x-hal`][] to point to this crate since this crate is actively maintained. Related issue: https://github.com/japaric/stm32f30x-hal/issues/36

[`awesome-embedded-rust`]: https://github.com/rust-embedded/awesome-embedded-rust#hal-implementation-crates
[`stm32f30x-hal`]: https://github.com/japaric/stm32f30x-hal

rubberduck203, 2020-01-04:
I created a PR on `awesome-embedded-rust` to address this.
https://github.com/rust-embedded/awesome-embedded-rust/pull/236

rubberduck203, 2020-01-04:
This is done.

Sh3Rm4n, 2020-01-04:
Should we wait to close this issue, until https://github.com/japaric/f3/pull/109
is merged? 

cc @dfrankland

Sh3Rm4n, 2020-01-04:
Thank you for taking action and submitting a PR to `awesome-embedded-rust` @rubberduck203 :+1:

rubberduck203, 2020-01-05:
@Sh3Rm4n I’m unsure what the PR on f3 has to do with this issue.

Sh3Rm4n, 2020-01-05:
Yeah, you are right, this is not directly related to this issue. I'll let @dfrankland decide to close this issue :)

dfrankland, 2020-01-06:
I think we are all good with the work @rubberduck203 put in for [`awesome-embedded-rust`][]. Thank you!

Considering the amount of documentation that links to [`stm32f30x-hal`][] I think it'll be good to get https://github.com/japaric/stm32f30x-hal/issues/36 closed as well. An example of this, is the [The Embedded Rust Book][] as of today, still points to [`stm32f30x-hal`][] (look under "Peripheral Access Crate (PAC)"):
<https://github.com/rust-embedded/book/blob/9f103c2a8aa91b959c7cddd31c950803d3ec06cc/src/start/registers.md>

I think I'll make a checklist at the top and start checking some of the items off.

[`awesome-embedded-rust`]: https://github.com/rust-embedded/awesome-embedded-rust#hal-implementation-crates
[`stm32f30x-hal`]: https://github.com/japaric/stm32f30x-hal
[The Embedded Rust Book]: https://github.com/rust-embedded/book

vringar, 2021-06-13:
The outdated crate is also linked in the [embedded-hal docs](https://docs.rs/embedded-hal/0.2.5/embedded_hal/index.html#suggested-implementation) and should probably be replaced there as well.