stm32-rs/stm32f3xx-hal #53: `downgrade()` results in hardfault if resulting array is used with non const index

wucke13, 2020-01-16:
This is a nifty bug and I'm not where it originates from, however, these were my observations so far:
```
let ld1 = gpiob.pb0.into_push_pull_output(&mut gpiob.moder, &mut gpiob.otyper);
let ld2 = gpiob.pb7.into_push_pull_output(&mut gpiob.moder, &mut gpiob.otyper);
let mut leds = [ld1.downgrade(), ld2.downgrade()];
```
The `downgrade()` call results in an hardfault, if and only if somewhere in the program there is an access to `leds`, where the index is not const/known at compile time.
No problem:
```
leds[0].set_high().unwrap();
for i in 0..2 {
  leds[1].set_high().unwrap();
}
```

Sure a problem (note: the hardfault does not occur on the `.set_high()` but on the `.downgrade()` above):
```
for i in 0..2 {
  leds[i].set_high().unwrap();
}
````
The same holds true for `for led in leds.iter_mut()` or `let mut i=0; loop {leds[i%2] ...; i+=1;}`

At this point, I have no clue how to debug any further. Any ideas?

dfrankland, 2020-01-17:
I do something similar in this example here, but I haven't seen the same issue:
https://github.com/dfrankland/proton-c/blob/c317202388425200c7da540fef0cbced7a68a848/examples/simple-keyboard/main.rs#L95-L110
https://github.com/dfrankland/proton-c/blob/c317202388425200c7da540fef0cbced7a68a848/examples/simple-keyboard/matrix.rs#L20-L41

wucke13, 2020-01-17:
Any reason why you call `.downgrade()` two times?
Side note: 
```
let mut leds = [ld1.downgrade().downgrade(), ld2.downgrade().downgrade()];
````
Doesn't work either.

wucke13, 2020-01-23:
Found my issue. My `memory.x` was wrong. Check your vendors manuals... 80K SRAM mean 64K SRAM + 16K CCRAM...

closed 2020-01-23 by wucke13