stm32-rs/stm32f3xx-hal #55: Missing code completion in IntelliJ

leon-jakob-schneider, 2020-01-28:
I know this may be resolved by an IDE plugin update in the future, but can we do something about this in the mean time?
[image]

It seams to me like the split function is not auto completed, because the IDE doesn´t find the macro generated struct `stm32f3xx_hal::gpio::gpioa::Parts`. Am I correct about this?

Sh3Rm4n, 2020-03-08:
I do not have any experience with IntelliJ. But as far as I know, macros are really hard to autocomplete. I know that rust-analyzer and the IntelliJ plugins are improving in a steady pace, so this issue will be resolved some time in the future. 

However, crate-wise we can do nothing about this. I'll close this issue, as this is off-topic. I hope you don't mind :)

closed 2020-03-08 by Sh3Rm4n

seanybaggins, 2021-01-07:
See https://github.com/rust-analyzer/rust-analyzer/issues/6029.