stm32-rs/stm32f3xx-hal #62: Module gpioe is not found

smedellin90, 2020-03-05:
`#![no_main]
#![no_std]

use embedded_hal::digital::v1_compat::{OldInputPin, OldOutputPin};
use rtfm::app;
use portable::button;
use stm32f3xx_hal::prelude::*;
use stm32f3xx_hal::{delay, gpio, i2c ,spi, stm32, timer};

#[cfg(not(test))]
extern crate panic_semihosting;

type Button0Pin = gpio::gpioa::PA6<gpio::Input<gpio::PullUp>>;
type BlueLED = gpio::gpioe::PE8<gpio::Output<gpio::PushPull>>;`

There is an error in type BlueLED. The module **gpioe** cannot be found within the **gpio** module. May someone please advise?

Sh3Rm4n, 2020-03-05:
Which microcontroller are you using / what feature did you enable in your Cargo.toml?

There is quite a bit of macro work involved creating the gpio modules depending on the feature. 

https://github.com/stm32-rs/stm32f3xx-hal/blob/609af0a91c2030ff2c3e35705e5394d66ec3cb81/src/gpio.rs#L1732

Here is a list of available features. The feature you choose has to match the microcontroller, for which you are planning to build it. 
https://github.com/stm32-rs/stm32f3xx-hal/blob/609af0a91c2030ff2c3e35705e5394d66ec3cb81/Cargo.toml#L51

smedellin90, 2020-03-05:
@Sh3Rm4n , this is my Cargo.toml:
`[package]
name = "embedded_testbench"
version = "0.1.0"
authors = ["Scott"]
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[workspace]

[dependencies]
stm32f3xx-hal = { version ="0.4.0", features = ["rt", "stm32f303"] }
##stm32f30x-hal = { version = "0.2", features = ["rt", "stm32f303" ] }
# = { version = "0.5", features = ["rt", "stm32f103" ] }
cortex-m = "0.6"
#cortex-m-rt = "0.6.3"
##cortex-m-rt = { version = "0.6", features = ["device"] }
cortex-m-rtfm = "0.5.0"
panic-semihosting = "0.5"
#bme280 = "0.2"
#pwm-speaker = { git = "https://github.com/TeXitoi/pwm-speaker", branch = "master" }
epd-waveshare = { version = "0.3.2", features = ["type_a_alternative_faster_lut"] }
embedded-hal = "0.2.1"
portable = { path = "portable" }

[profile.dev]
opt-level = 1
lto = true
incremental = false

[profile.release]
lto = true
incremental = false
opt-level = "z"
debug = true`

I noticed that the device "stm32f303" does not contain the gpioe module, which would explain the errors. What is the difference between stm32f303 and the rest of the stm32f303x**X** devices? I currently cannot find a good explanation online. The physical board I am using is the STM32F3Discovery (STM32F303VCT6)).

Sh3Rm4n, 2020-03-05:
Okay, this seems odd, I'll have to try to reproduce this myself. 

The other stm32f303.. have some slight differences in the peripherals, e.g the usb-peripheral iirc. Otherwise, they should be the same. In this case, it shouldn't matter. As you can see, if you choose one of these, they enable the `stm32f303` feature.

strom-und-spiele, 2020-03-06:
@smedellin90 
If you align it this way you'll probably see what to set for the **`X`**:
`stm32f303xX`
`stm32f303vct6`
I highly recommend specifying the `stm32f303xc` feature as it allows for more flexibility and peripheral use.

Also (as I ran into this earlier): make sure the Pins you are using are not connected to some internal IC on your discovery board as this might mess with your input/outputs.

hth & hf :)

smedellin90, 2020-03-06:
@Sh3Rm4n  and @strom-und-spiele ,

I decided to try out stm32f303xb and write a simply blinky program and it worked. 
I'm new to rust, as well as embedded software, so I truly appreciate your answers. Thank you for taking the time!

-Scott

closed 2020-03-06 by smedellin90

Sh3Rm4n, 2020-03-08:
Okay, I took another look at it. GPIOE is explicitly excluded to build for feature `stm32f303` as the alternate functions differ between the variant `stm32f303xb`, `stm32f303xc` etc.