stm32-rs/stm32f3xx-hal #66: Support DMA

Sh3Rm4n, 2020-03-08:
Direct Memory Access allows peripherals to write to memory directly, without needing buffers. This saves memory and is a handy feature.

ra-kete, 2020-03-31:
I'm interested in this too. The project I'm working on pretty much demands DMA (in tandem with ADC and SPI). That's a major reason I don't use this HAL for it but wrote [my own one](https://gitlab.com/ra_kete/stm32f303xc-hal), based on [japari's blog post](https://blog.japaric.io/safe-dma/). I planned to upstream that at some point, but wasn't really sure how to go about it. My implementation is pretty specific to my needs and probably needs to be refactored a bit to be generally useful. And of course it supports only STM303xC MCUs.

I'd be willing to try my hand on implementing something that's useful for everyone. If you don't already have other plans, of course :)

Sh3Rm4n, 2020-04-01:
Cool. I have no plans to implement this myself atm. So I'm looking forward for any support on this. 

I'm okay with it being only compatible with the stm32303xc for now (similar to https://github.com/stm32-rs/stm32f3xx-hal/pull/47) If it is working, we can always incrementally add support for more stm32f3xx chips later in followup PRs. :)

ra-kete, 2020-04-05:
I've created a PR that adds MVP DMA support, i.e. the minimum amount of functionality that's useful to actually do something. I think it's a good idea to add more functionality incrementally instead of having one huge PR in the end that nobody understands and that takes forever to get merged.

Adding the missing DMA features I can think of:

- [x] generic support for one-shot DMAs
- [ ] generic support for circular DMAs
- [x] support for serial DMA
- [ ] support for SPI DMA
- [ ] support for I2C DMA
- [ ] support for ADC DMA (blocked by #20)

ra-kete, 2020-04-05:
Here are a few references I found helpful for DMA implementation:

- [The Embedonomicon on DMA](https://docs.rust-embedded.org/embedonomicon/dma.html)
- [`embedded-hal` issue about DMA](https://github.com/rust-embedded/embedded-hal/issues/37)
- [DMA Implementation in `stm32l0xx-hal`](https://github.com/stm32-rs/stm32l0xx-hal/blob/master/src/dma.rs)