stm32-rs/stm32f3xx-hal #81: STM32F303 - GPIOE.split() missing

bobgates, 2020-03-23:
I'm trying to get something running on the STM32F3Discovery board, which has an STM32F303 processor.

I've hit a problem where ```cargo build``` reports:

```
error[E0599]: no method named `split` found for type `stm32f3::stm32f303::GPIOE` in the current scope
  --> src/main.rs:49:38
   |
49 |         let mut gpioe = device.GPIOE.split(&mut rcc.ahb);
   |                                      ^^^^^
```

What's odd about this is that if I change the port to any of GPIOA-GPIOF other than GPIOE, there's no problem. And if I change the hal feature to "stm32f373" instead of "stm32f303" there's also no problem. 

Any ideas?

Sh3Rm4n, 2020-03-23:
Thanks for transferring @adamgreig :) 

Yes this is indeed a HAL issue. 
No issue directly, but a confusing way about the device features. 

`stm32f303` is a least common functionality subset of the subvariants `stm32f303xb`, `stm32f303xb`, `stm32f303xb`, `stm32f303xb`, `stm32f303x6` and `stm32f303x8` and as `stm32f303x6` and `stm32f303x8` are missing the gpioe peripheral, it is not included for `stm32f303`. 

It should work, when you use the `stm32f303xc` feature, which is the microcontroller on the stm32f3 discovery board. 

This confusen happend before as well: 

https://github.com/stm32-rs/stm32f3xx-hal/issues/62

We do plan to change this in the future, probably by removing the `stm32f303` feature, so that this confusion does not happen in the future. 

https://github.com/stm32-rs/stm32f3xx-hal/pull/75

bobgates, 2020-03-23:
Thanks for the clarity. Going up and down the Arm-core stack is quite confusing for beginners!

Sh3Rm4n, 2020-04-18:
The `stm32f303` feature is now removed as it is to u specific, see #75.

closed 2020-04-18 by Sh3Rm4n