stm32-rs/stm32f3xx-hal #83: External interrupts (example, question)

jkboyce, 2020-04-02:
I wrote a simple example of GPIO-driven interrupt handling, available [here.](https://github.com/jkboyce/stm32f3xx-hal/blob/master/examples/gpio_interrupt.rs) Compile with `cargo build --example gpio_interrupt --features "rt stm32f303xc"` On the F3 Discovery board it toggles an led when you push the user button.

I'm curious if anyone's working on–or has thoughts about–adding better support for interrupts. The [`gpio.rs`](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/gpio.rs) in `stm32f4xx-hal` has some nice methods based on the `ExtiPin` trait, making it easy to set up interrupts without direct register access. For example:

```rust
let gpioc = dp.GPIOC.split();
let mut board_btn = gpioc.pc13.into_floating_input();
board_btn.make_interrupt_source(&mut dp.SYSCFG);
board_btn.enable_interrupt(&mut dp.EXTI);
board_btn.trigger_on_edge(&mut dp.EXTI, Edge::RISING);
```

It would be nice to get the f3/f4 HALs into closer alignment but I wanted to gauge others' thoughts before jumping in.

Sh3Rm4n, 2020-04-02:
I'm all in for being more aligned with the f4 HAL as these chips are similar in a lot of ways anyway. 

I've had no time to do this myself, so I'm happy for any PR for this :) 

Seems quite useful. Using interrupts myself in a project of mine, I needed some time to configure them right. These methods would be the process much easier and more discoverable.

David-OConnor, 2020-06-22:
I'm looking forward to this eventually being fixed. I don't understand low-level/unsafe code enough to impl myself confidently, but jk - thank you for the detailed, commented example.

Sh3Rm4n, 2020-07-18:
> Have you had any luck configuring timer-based interrupts? Like a delay, but using a low-power mode? I suspect it'd be similar to your GPIO interrupt example, but use a timer to trigger the wakeup. Is this still something we'd use a WFI, or would it be a WFE? Thank you. And do you have an example of how to set it up for other GPIO pins? We presumably modify the address at the end of this line:

I've not experimented with timer based interrupts, but technically this should be possible. It really depends on your platform implementation and if you want to implement an interrupt event based system.
According to the [ARM Documentation](https://developer.arm.com/documentation/dui0489/e/arm-and-thumb-instructions/miscellaneous-instructions/sev--wfe--wfi--and-yield) `WFE` is just an extension of `WFI` which also resumes from suspension on some more special events (`multicore` / `mulitrhreading`).

David-OConnor, 2020-07-18:
Thank you for the details. After more research and generous help from others, including this thread's author, I've made progress on the issue: It seems the best way to do this is fit the chip with a 32.768k crystal, enable the Low Frequency External oscillator, and set up the `RTC Wakeup Timer` ,which will count down, then fire an interrupt.

Doing this requires a number of steps, which are outlined in the Ref manual, and other ST documentation. I will post the result as a PR or external crate once complete.

PR #189 (2021-03-27): GPIO refactor (Simplify macro, reduce type states, and support interrupt)
* Simplify macro
  `gpio.rs` uses more generics than macro now
* Collapse Input type arguments
  An attempt to address #125
* Support interrupt
  Close #83 and #112
closed 2021-03-27 by Sh3Rm4n