stm32-rs/stm32f3xx-hal #84: Implement InputPin trait for PA1<Output<OpenDrain>>

peter-wilkins, 2020-04-02:
I'm trying to read a dht11 humidity sensor with a stm32f303 discovery board

the example in the dht11 crate is for stmf0xx boards:
https://github.com/michaelbeaumont/dht-sensor/blob/master/examples/stm32f042.rs#L30

Rather than give in and just buy an stm32f0xx board I'd like to have a crack at adding this trait. If anyone knows that it is futile (eg  `Output<OpenDrain>` is not both `InputPin` and `OutputPin`  for the f3) or not useful in anyway please say so.

Part-converted example: https://gist.github.com/peter-wilkins/b5c914b1f8f5b4098b198133b4280472

I'm just starting out with embedded but will attempt to put together an PR, it may take some time. Any advice appreciated.

Sh3Rm4n, 2020-04-04:
The hardware design of the gpio's from the stm32fx-line should all be the same, so nothing speaks against adding the `InputPin` trait to `Output<OpenDrain>`. And it is clearly missing right now in this HAL. Seems like that besides stm32f0xx, stm32f1xx and stm32f4xx have these traits as well. 

So if you are willing to put together a PR, go for it. I appreciate every contribution. :) 

You could take the implementation from the [stm32f4xx-hal](https://github.com/stm32-rs/stm32f4xx-hal/blob/ddccb4fc48908f66f3848e686e87961a0df73a43/src/gpio.rs#L699). 

[stm32f1xx-hal](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/gpio.rs) has a similar solution, but only implements it for OpenDrain. I don't know if we should limit the Trait to `Output<OpenDrain>`. I tend towards the more generic solution from the stm32f4xx-hal :)

David-OConnor, 2020-06-01:
I think I'm running into this as well, in an attempt to get the DS18B20 temp sensor working.

David-OConnor, 2020-06-02:
https://github.com/stm32-rs/stm32f3xx-hal/pull/105

therealprof, 2020-06-19:
> You could take the implementation from the [stm32f4xx-hal](https://github.com/stm32-rs/stm32f4xx-hal/blob/ddccb4fc48908f66f3848e686e87961a0df73a43/src/gpio.rs#L699).

The problem with that is that it will raise wrong expectations and potentially cause hardware failure through incorrect use. You cannot safely do this with a pin in push-pull mode, guess we should have been more careful in the review of that change. 😅

David-OConnor, 2020-06-19:
So... How can we get the one wire protocol working?

David-OConnor, 2020-06-22:
The [onewire](https://github.com/kellerkindt/onewire) crate released an unrelated patch on Github, which now works with this lib... if you use my Open PR.

I guess re reluctance to merge this: What alternate do you suggest to get protocols like this working? This one in particular is popular in some circles.

PR #114 (2020-07-18): Implement InputPin for Output<OpenDrain> pins
Forked from #105 and added some cleanup.

Resolves #84.
Once this is merged we can close both #88 and #105.
closed 2020-07-18 by ra-kete