stm32-rs/stm32f3xx-hal #87: Initialization of real time clock on stm32f3 Discovery board

smedellin90, 2020-04-06:
Hello! 

I have not been able to find a way to initialize the rtc on the  stm32f3 Discovery board.
One the STM32F103C8, initialization would go something like this (using cortex-m-rtfm crate):

`
 #[init(spawn = [msg])]
 fn init(mut c: init::Context) -> init::LateResources {
     let mut rcc = c.device.RCC.constrain();
     let mut backup_domain = rcc
          .bkp
          .constrain(c.device.BKP, &mut rcc.apb1, &mut c.device.PWR);
     let mut rtc_dev = rtc::Rtc::rtc(c.device.RTC, &mut backup_domain);
     // rest is omitted
}
`
Is there a way to initialize the real time clock currently using the stm32f3xx-hal?

Sh3Rm4n, 2020-04-06:
Currently, the RTC module is missing from the stm32f3xx-hal. But the stm32f3-series do support RTCs, so it should be possible. 

PRs for an [rtc](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/rtc.rs) module are welcome! :)

smedellin90, 2020-04-10:
I'm currently working on an RTC module for the stm32f3xx-hal. I'm running into a peculiar issue at the moment with the RTC init flag (RTC_ISR register, INITF bit).  Below is a function that writes prescaler values into the PRER register:

```
    fn perform_prescaler_write(&mut self, prediv_s: u32, prediv_a: u32) {
        // Disabling write protection to RTC reg block
        self.regs.wpr.write(|w| unsafe { w.bits(0xCA) });
        self.regs.wpr.write(|w| unsafe { w.bits(0x53) });
        // Setting RTC Initialization bit to make prescalers programmable
        self.regs.isr.modify(|_, w| { w.init().set_bit()});
        // wait for initilization mode to take effect
        while self.regs.isr.read().initf().bit() == false {} // Never finishes
        // Setting Prescalers for 1 hz RTC
        let raw_bits: u32 = prediv_s | (prediv_a << 16);
        self.regs.prer.write(|w| unsafe { w.bits(raw_bits) });
        // Take device out of Initialization mode
        self.regs.isr.modify(|_, w| { w.init().clear_bit()});
        // wait for last write to be done
        while !self.regs.isr.read().initf().bit() == false {}
    }
```
It will get hung up, waiting for RTC initialization mode to be set. Has anyone run into this issue? I'm currently using the stm32f3 discovery board.

David-OConnor, 2020-07-15:
@smedellin90 : Any luck on the RTC module? I'm attempting to do the same, for RTC interrupts.

David-OConnor, 2020-09-15:
Is this working now?