stm32-rs/stm32f3xx-hal #103: How do you use Delay in multiple places

David-OConnor, 2020-05-16:
Let's say you have something that consumes the delay:

```rust
    let mut cp = cortex_m::Peripherals::take().unwrap();
    let dp = stm32::Peripherals::take().unwrap();

    let stim = &mut cp.ITM.stim[0];

    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);
    let mut delay = Delay::new(cp.SYST, clocks);
    let mut lcd = HD44780::new_i2c(i2c, LCD_I2C_ADDRESS, delay);
```

How do you then use a delay in the loop? It looks like the `Delay::free` method may be along the right tracks, but I'm not sure. Tried re-initializing `cp` and `delay` with no luck, and it appears you can't clone either, or use as a ref.

Sh3Rm4n, 2020-05-18:
I don't know the details about the HD44780 trait, but I assume that it consumes delay. 

Maybe HD44780 provides a method to release delay again?

`Delay::free` does the same for `systck`. It consumes it and to give it back, there is a `free` method. 

This is a common pattern used in embedded rust to move peripherals around, when references are not an option.

David-OConnor, 2020-05-18:
It sounds like you're right, that the answer is for libs not to consume the delay. I submitted a (merged) PR to the lib so that it accepts mutable refs in methods, vice consuming the delay as a field.

closed 2020-05-23 by David-OConnor