stm32-rs/stm32f3xx-hal #104: OneWire support

David-OConnor, 2020-06-01:
OneWire is a digital protocol used by a very popular temperature sensor, the DS1820B. I'm unable to get it working with this crate, and am wonder what you think the best approach is.

I'm suspicious [this recent issue](https://github.com/stm32-rs/stm32f3xx-hal/issues/84) is related.

I think the issue is that the GPIO pins this library uses are generally input or output only, and don't implement both. Is this correct? The specific issue I'm running into with [this OneWire crate](https://github.com/kellerkindt/onewire/blob/master/src/lib.rs) is that it expects the pin to implement this:
```rust
pub trait OpenDrainOutput<E>: OutputPin<Error = E> + InputPin<Error = E> {}
```.

Is this an issue of implementing both `InputPin` and OutputPin` for gpio pins configured as open drain? I looked through [this source file](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/gpio.rs) to try to find a solution, either robust, or a hack to make OneWire work, but am not sure how to deal with the macros.

Thoughts? Is this the same thing as the issue I linked near the top?

ra-kete, 2020-07-18:
@David-OConnor I believed this was fixed with merging #114. Can you confirm?

David-OConnor, 2020-07-19:
Affirm. Closing!

closed 2020-07-19 by David-OConnor