stm32-rs/stm32f3xx-hal #110: Problem matching STCubeMX RCC configuration

nicolai86, 2020-07-03:
I've having trouble migrating my STCubeMX RCC configuration to a working configuration with `stm32f3xx-hal`.

I'm using a STM32F303CCT6 with 16 MHz HSE and USB. The clock configuration works fine for USB-Serial when generating a project via STCubeMX:

[image]

Now I'm trying to convert attached clock configuration, but this fails the usb validation it seems:

```
let clocks = rcc
        .cfgr
        .use_hse(16.mhz())
        .hclk(72.mhz())
        .sysclk(72.mhz())
        .pclk1(36.mhz())
        .pclk2(72.mhz())
        .freeze(&mut flash.acr);
assert!(clocks.usbclk_valid());
```

when I switch to 48 mhz it works fine:

```
let clocks = rcc
        .cfgr
        .use_hse(16.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .freeze(&mut flash.acr);
    assert!(clocks.usbclk_valid());
```

Is this a problem with the dynamic pll_config calculation, or am I missing something?

Sh3Rm4n, 2020-07-20:
Hey! Sorry for the late answer. 

> Is this a problem with the dynamic pll_config calculation, or am I missing something?

Most likely due to the dynamic pll_config

Are you using the current master or the latest version (0.4.2), and in the latter case, can you try it with current master again?
I am asking because I recently merged a changed in https://github.com/stm32-rs/stm32f3xx-hal/pull/67, which changes the calculation quite a bit and should match the cubemx configuration more closly, as it is utilizing the PLL deviders now. 

I'm curious, if I have fixed or introduced a bug :grinning:

nicolai86, 2020-07-21:
@Sh3Rm4n I used v0.4.3, which failed to work with above configuration. switching to master makes it work. Looking forward to the next release :)

closed 2020-07-21 by Sh3Rm4n