stm32-rs/stm32f3xx-hal #111: Apoint more maintainers

ra-kete, 2020-07-13:
Lately this project seems to be quite undermaintained. There are a couple of PRs open currently that would be useful to have merged, but they don't get reviewed.

I think that is a problem, since the stm32f3xx-hal is usually how beginners in embedded Rust (like me) have their first contact with the ecosystem (due to the great Discovery book). Bugs and missing features of this crate can lead to needlessly steep learning curves, so it would be good if we could fix those in a timely manner.

I am sure the current maintainers have a busy life and I certainly don't want to blame anyone! But I think we need to find a solution to this problem. The most obvious one would be giving maintainer rights to more people, to share the load of getting PRs reviewed and merged.

dfrankland, 2020-07-13:
I agree! Folks who'd like to help maintain. Please, feel free to reach out to me and I can get you setup 👍

Sh3Rm4n, 2020-07-14:
Thanks for taking action! 

I'm really sorry for being silent in the last few months without notice. I was busy with my internship and quite overwhelmed by the PR's and took a step back. 

I'm a rust beginner myself and I lack the experience too confidently guide the current PR's to the right design decision. In the end I always looked at [stm32f1xx-hal] or [stm32f4xx-hal] how they've done it, because there is where the innovation takes place. I haven't had the time to give constructive feedback besides nitpicking on some points. 
It ended up that if I had the spare-time and motivation to do some embedded rust, I felt the pressure to take care of this repository (as many are still waiting on my feedback). But I really lack the experience to help productively, as I'm missing the time to play around with this stuff myself. 

If more people would work on the repository, it would help enormously

> I think that is a problem, since the stm32f3xx-hal is usually how beginners in embedded Rust (like me) have their first contact with the ecosystem (due to the great Discovery book). Bugs and missing [...]

Strongly agree on that point! There are some points I would like to improve to smoothen the experience for beginners. I'd wish for better alignments between the stm32-rs hal's on implementation design for some modules for more consistent interfaces overall,
(E.g [this](https://github.com/stm32-rs/stm32f3xx-hal/issues/37) issue comes to my mind). Some "does" and "don'ts" or rather best practices would help. I think focusing on the API design is the most important. Implementation details can always be improved with time.
But this could be discussed in the new [stm32-rs/meta](https://github.com/stm32-rs/meta) repo to open this discussion up to everyone.

I also think that II'll be able to get some work done on some smaller PRs and issues in the next few days again. :)

ra-kete, 2020-07-14:
@Sh3Rm4n No worries! And thanks for the effort you have put in so far!

@dfrankland has given me contributor rights to this repo now too, so I will be able to help out in the future. Having more people would be good of course. I can't (or rather: shouldn't) review my own PRs ;)

seanybaggins, 2021-09-26:
Hi @Sh3Rm4n, @dfrankland, and @ra-kete . I am currently in the job market. Hoping to get a job that is remote, works with embedded, and utilizes Rust. I would love to help out with maintenance while I am looking for job.

Sh3Rm4n, 2021-10-04:
Hey @seanybaggins, I'm happy you are interested to help and wanting to contribute to stm32f3xx-hal! :) 

I sadly had not much time in the last weeks to make progress on some PRs I'm working on, so many of these are still WIP. And there are also different issues, which needs to be tackled. 

But I suggest you just go through these, find things you are interested in solving or even find a peripheral, which is currently not yet supported and draft possible implementations. 

You can always open PRs, and I'll try my best to review them. And you can ask me anything, if you have any questions :)