stm32-rs/stm32f3xx-hal #112: Interrupt support

David-OConnor, 2020-07-14:
I'd like to see this library include high-level code to handle interrupts, and am curious how y'all would like to approach this. @jkboyce  , who posted [this issue](https://github.com/stm32-rs/stm32f3xx-hal/issues/83) has been immensely helpful, as have several people on the `stm32` Matrix chat, especially `adamgreig`.

I've been working off a standalone module. Here are three possible approaches:
1 - Achieve API and implementation parity with other stm32 hal crates. I haven't done my research on this, although understanding how this API is set up isn't trivial due to heavy use of macros and traits. . API parity has obvious advantages, so this is the ideal solution.

2 - Add a standalone module to this crate using hand-written, repetitive fns

3 - Same as above, but using macros to make it cleaner to maintain. I'm not sure how to do this, as my trivial attempts have failed. Might need to use the [paste crate](https://crates.io/crates/paste) or proc macros.

4 - As a standalone module released on `crates.io`. This is the backup approach. I would never have been able to get interrupts working in my crate without the help of the two people listed at the top of this post, and want this to be a solved problem going forward.

Here are the basics for GPIO interrupts, using option 2. It works, but suffers from DRY. The remaining interrupt functionality (The 4 types of RTC interrupts, I2c interrupts etc) should be added as well, either in the initial PR, or later.

```rust
/// Reference the STM32F303 reference manual, section 14.2.5 for a `Functional description`
/// of the steps we accomplish here.
/// 0 - 15.
pub fn setup_line(
    exti: &mut EXTI,
    line: u8,
    task: Task,
    edge: Edge,
) {
    // Configure the Trigger Selection bits of the Interrupt line (EXTI_RTSR and EXTI_FTSR)
    let rise_trigger;
    let fall_trigger;
    match edge {
        Edge::Rising => {
            // configure EXTI$line to trigger on rising edge, disable trigger on falling edge
            rise_trigger = true;
            fall_trigger = false;
        }
        Edge::Falling => {
            // configure EXTI$line to trigger on falling edge, disable trigger on rising edge
            rise_trigger = false;
            fall_trigger = true;
        }
    }
    match line {
        0 => {
            // Configure the corresponding mask bit in the EXTI_IMR register.
            exti.imr1.modify(|_, w| w.mr0().unmasked());
            exti.rtsr1.modify(|_, w| w.tr0().bit(rise_trigger));
            exti.ftsr1.modify(|_, w| w.tr0().bit(fall_trigger));

            // Configure the enable and mask bits that control the NVIC IRQ channel mapped to the
            // EXTI so that an interrupt coming from one of the EXTI line can be correctly
            // acknowledged.
            unsafe { NVIC::unmask(interrupt::EXTI0) };
        }
        // ....
        5 => {
            exti.imr1.modify(|_, w| w.mr5().unmasked());
            exti.rtsr1.modify(|_, w| w.tr5().bit(rise_trigger));
            exti.ftsr1.modify(|_, w| w.tr5().bit(fall_trigger));
            unsafe { NVIC::unmask(interrupt::EXTI9_5) };
        }
        // ...
    }
}

/// Configure a GPIO to work with an interrupt line.
/// The `line` parameter corresponds to both the interrupt line, and the
/// GPIO number. Eg, `3` means `line3`, and `pb3`.
/// Reference section 12.1.3 of the datasheet for a breakdown by CR
/// and line.
pub fn setup_gpio(syscfg: &mut SYSCFG, gpio_reg: GpioReg, line: u8) {
    // Select this GPIO pin as source input for EXTI$line external interrupt

    match line {
        0 => syscfg
            .exticr1
            .modify(|_, w| unsafe { w.exti0().bits(gpio_reg.cr_val()) }),
        1 => syscfg
    // ...
    }
}

/// Set EWUP bit in PWR_CSR register; set PWREN bit to enable power interface clock.
/// This is required to enable use of the WKUP pin, eg to wake up from standby.
/// https://vasiliev.me/blog/sending-stm32f1-to-deep-sleep-with-rust/#fn:3
/// We use our custom stm32f3xx hal branch to allow access to the (normally private) `enr`.
pub fn setup_wakeup(apb1: &mut APB1, pwr: &mut PWR) {
    // todo: We must somehow call this too:
    // enable SYSCFG clock to enable external interrupts; must come before RCC.constrain()
    // dp.RCC.apb2enr.write(|w| w.syscfgen().enabled());

    apb1.enr().modify(|_r, w| w.pwren().set_bit());
    pwr.csr.modify(|_, w| w.ewup1().set_bit());
}

/// These functions are called when their appropriate interrupt line
/// is triggered.
/// Eg:
///     make_interrupt_handler!(EXTI3);
///     make_interrupt_handler!(EXTI9_5);
///     make_interrupt_handler!(EXTI15_10);
macro_rules! make_interrupt_handler {
    ($line:ident) => {
        #[interrupt]
        fn $line() {
            free(|cs| {
                // Reset pending bit for interrupt line
                // todo: pr1 for all?
                unsafe { (*EXTI::ptr()).pr1.modify(|_, w| w.pr1().bit(true)) };
                // hprintln!("In exti $line").ok();
            });
        }
    };
}
```

Example use:
```rust
   // Setup an interrupt that activates when `Pb3` is pulled low.
    interrupt_::setup_line(
        &mut dp.EXTI,
        3,
        interrupt_::Task::Enable,
        interrupt_::Edge::Falling,
    );
    interrupt_::setup_gpio(&mut dp.SYSCFG, interrupt_::GpioReg::B, 3);
    interrupt_::setup_wakeup(&mut rcc.apb1, &mut dp.PWR);
    make_interrupt_handler!(EXTI3);
```

ra-kete, 2020-07-14:
I too think option 1 would be the preferred solution. @Sh3Rm4n also mentioned as much in the comments of #83. If we can just take the stm32f4xx-hal implementation and tweak it according to the different hardware, that should work well as a first implementation. I haven't looked at the details, but since you also list other options, do you see any issues with option 1?

Btw, you are right that the HALs (especially the GPIO abstractions) are full of macro soup, which makes the code hard to read at times. Sometimes it could be done better, sometimes not so much. But usually the macro way is the easiest way to implement the HAL logic. And at some point you get used to reading it too. It's still always good to try and think how we can improve the situation and the paste crate is definitely something that can help here.

David-OConnor, 2020-07-14:
I mainly haven't dug into option 1; Ie my approach was *get it working*, then finer points.

David-OConnor, 2020-07-15:
Here's [an example](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/stopwatch-with-ssd1306-and-interrupts.rs) using a GPIO interrupt from f4:
```rust
        // Create a button input with an interrupt
        board_btn.make_interrupt_source(&mut dp.SYSCFG);
        board_btn.enable_interrupt(&mut dp.EXTI);
        board_btn.trigger_on_edge(&mut dp.EXTI, Edge::FALLING);
```
The API looks good, but I haven't looked at the implementation.

jkboyce, 2020-07-15:
Thanks for teeing up options @David-OConnor . Personally I like option 1, since the gpio interrupt support in stm32f4xx-hal is pretty good, and having a similar/identical API would greatly simplify porting code between f3 and f4 chips (one of the presumed benefits of a HAL layer in the first place).

See for instance stm32f4xx-hal's [gpio.rs](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/gpio.rs) and its <code>ExtiPin</code> trait (the f1 hal works the same way). Unfortunately the macros are quite different from the stm32f3xx-hal ones and so it would be a significant re-plumbing to get them aligned. But it would be a great service!

David-OConnor, 2020-07-16:
After some review, I'm not up to the task of elegantly adding this in a way that achieves API parity with stm32f4. Based on jkboyce's last post, it sounds like the ideal approach would be a refactor of the macro system.

Would y'all be down to merge Approach #2 (standalone module to go with this crate) if I PR it? If not, I'll go with #4 as a stopgap.

Sh3Rm4n, 2020-07-18:
> See for instance stm32f4xx-hal's gpio.rs and its ExtiPin trait (the f1 hal works the same way). Unfortunately the macros are quite different from the stm32f3xx-hal ones and so it would be a significant re-plumbing to get them aligned. But it would be a great service!

Our `gpio!` macro system evolved into being much more verbose in comparison to  stm3wf4xx-hal's implementation, but it is generally the same thing. And add first glance the `exti!` macro is just an extension on top of it. I don't see a technical problem adding this to this HAL as well. 

In the end, i really like there API, as it is pretty [simple](https://github.com/stm32-rs/stm32f4xx-hal/blob/8f9aaf82e9104f33d4dcf3d1c16da5c51c5361ce/src/gpio.rs#L86-L94) and comes downt to:

```rust
/// External Interrupt Pin
pub trait ExtiPin {
    fn make_interrupt_source(&mut self, syscfg: &mut SYSCFG);
    fn trigger_on_edge(&mut self, exti: &mut EXTI, level: Edge);
    fn enable_interrupt(&mut self, exti: &mut EXTI);
    fn disable_interrupt(&mut self, exti: &mut EXTI);
    fn clear_interrupt_pending_bit(&mut self);
    fn check_interrupt(&self) -> bool;
}
```

I don't think it can get much more simple, as this interrupt configuration of ST is the way it is.

David-OConnor, 2020-07-19:
I can post the steps to execute (A continuation of my above example) for GPIO interrupts, if someone can translate it into macros. Steps for RTC wakeup's a WIP. It's more complicated, but still can be reduced to a high-level API

Sh3Rm4n, 2020-07-28:
This looks like an interesting approach. I briefly looked at your implementation, but I'll take another closer look soon to leave my comments about it :)

David-OConnor, 2020-07-28:
Great! I've still been unable to get RTC interrupts working, and haven't attempted other types yet. Won't be able to touch this again for about 2.5 weeks. Will keep working off my module until further notice.

David-OConnor, 2020-08-17:
It looks like [this HAL crate](https://github.com/stm32-rs/stm32l0xx-hal/blob/master/src/exti.rs) uses the module approach.

seanybaggins, 2021-01-24:
I was thinking about going through and refactoring the macros with GPIO interrupts. I want to use proc macros to implement this. Is there any gotchas I should be aware of? I noticed that if we try to make the gpio proc macros a sub crate of the hal, cargo will try to use the target settings of the hal for compiling the macros which is not what we want.

Here is a rough outline of what I was hoping the macros to look like.
```rust
// fully_erased
// Should not be a macro. It should just be implemented once.
// Should GPIO also be a trait that is implemented across all crates?
//     More or less I think it already is. Just in smaller form factors such as

/// gpio_partially_erased!
/// Generates a partially erased gpio pin.
/// # Example
/// ```rust
/// use gpio_macros::gpio_partially_erased;
/// gpio_partially_erased!(PAx);
/// gpio_partially_erased!(PBx);
/// //--snip--
/// gpio_partially_erased!(PFx);
/// ```

// gpio_pin!
// Generates the code for a gpio with a pin number
// # Example
// ```rust
// gpio!(PA1);
// gpio!(PA2);
// // --snip--
// gpio!(PF10)
// ```
```

David-OConnor, 2021-01-24:
Great. I've been using the standalone module in my fork, but would prefer the GPIO method for consistency with other HALs

seanybaggins, 2021-01-25:
Question. I am looking at the [stm32f4xx_hal documentation]( https://docs.rs/stm32f4xx-hal/0.8.3/stm32f4xx_hal/gpio/gpioa/struct.PA0.html). They have every alternate function implemented for each pin even though according to [this documentation](https://www.st.com/resource/en/datasheet/stm32f429i.pdf) each pin does not support every alternate function.

I think this may be a reasonable spot to deviate from stm32f4xx-hal. Thoughts?

Also, I have noticed the following patterns that will allows us to simply the API if we use proc_macros instead of declaritive. Those patterns are...
```
afr = {
    H/h if pin# >= 8,
    L/l otherwise
}

reset = {
    AF0 if (portA and pin# in Range(13..=15)) or (portB and pin# in Range(3..=4)),
    INPUT<FLOATING> otherwise
}
```

This allows us to determine a lot of information just from the port letter and the pin# and allows the macro api to simplified to something of the following.
```rust
gpio_partially_erased!(PAx)
gpio_pin!(PA0, af: [1, 3, 7, 8, 9, 10, 15])
```

Do people feel comfortable if this was to become the gpio macro API? Is it okay that it deviates from what is auto created from `codegen`?

Sh3Rm4n, 2021-08-16:
Interrupt support, at least for gpios was added in #189

closed 2021-08-16 by Sh3Rm4n