stm32-rs/stm32f3xx-hal #125: Collapse Input<OpenDrain / PullUp / PullDown) into Input

mattiasgronlund, 2020-07-26:
### Rethink Pin Type States in Regard of Pull Up/Down 

Currently Input<> has three different states (Floating, PullUp, PullDown) which expose the pull-up and pull-down settings.
But Output / AF has no such distinction, instead there is a method internal_pull_up(bool) available for AF and Output<OpenDrain>, which make it possible to switch between floating of pull-up.
I see a couple of problems with this model:
1. it is not consistent
2. it limits the available settings more than the MCU really does, which might be a problem in some situations.
3. I would call the current implementation as being buggy, as it currently allows one to fool it. You can move a pin into Input<PullDown> then move it to Output<OpenDrain> or even Output<PushPull> and as neither of the later moves resets the pull up/down states you enter a state which should have been "blocked".

My suggestion is that we collapse the Input<*> states into just Input.

One other solution would be to actually add more type states, but I think that would be a bad idea as the pull up/down is more of a hardware choice to have in the MCU and not external on the board.

Sh3Rm4n, 2020-07-28:
[`Floating, PullUp, PullDown`](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/gpio.rs#L30-L35) for Input is, what [`OpenDrain, PushPull`](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/gpio.rs#L42-L45) is for Output. 

> One other solution would be to actually add more type states, but I think that would be a bad idea as the pull up/down is more of a hardware choice to have in the MCU and not external on the board.

The stm32f4xx-hal somewhat recently added a [`OpenDrain`](https://github.com/stm32-rs/stm32f4xx-hal/pull/122/files) implementation for AFs. So this seems like being the trend for the other HALs, at least for Output / AF pins. 

> My suggestion is that we collapse the Input<*> states into just Input.

I see that this is a viable solution. I can't really think of a situation, where it is needed to encode the pull state in a typestate and as you said, an Input pin can be `Floating` internally, but still a pull-up resistors is attached externally. For consistency, there should be a method to transfer the Input from `Input<Floating>` to `Input<Pullup>` without activating the internal pull-up, which seems strange. 

Point **3.** seems rather important though. I would classify this as an implementation bug we have to fix.

Sh3Rm4n, 2020-12-10:
> For consistency, there should be a method to transfer the Input from Input<Floating> to Input<Pullup> without activating the internal pull-up, which seems strange.

Regarding this point: I think to leave it as is, where there is only one way to convert `Input<Floating>` to `Input<Pullup>`, which then enables the internal pull up. First and foremost this crate provides an abstraction over the microcontroller itself. 

When a user needs a `Input<Pullup>` where an external input is used instead of an internal pullup, they should use and `Inpuut<Floating>` and maybe wrap it around a newtype. This more closely resembles the state of the gpio peripheral inside the MCU.

Piroro-hs, 2020-12-16:
IMHO, 4 GPIO modes (`Output<OTYPE>`, `Alternate<AFx, OTYPE>`, `Input` and `Analog`, where `OTYPE` is `OpenDrain` or `PushPull`) and 2 methods (`set_internal_pull_up_down()` and `set_output_speed()`) is more consistent and reflect MCU capability.

Sh3Rm4n, 2021-03-28:
This should be closed by #189 I think :)

closed 2021-03-28 by Sh3Rm4n