stm32-rs/stm32f3xx-hal #131: Pwm input mode in timera

talalahsan01, 2020-08-08:
Hey! I am working on something where i needed the timer pwm input mode and i didn't find anything in the hal to implement this as in this crate there is a way to implementation pwm but pwm input is not available. I want to get the duty cycle and frequency of a pwm signal being input to the board and i also it needs to be 4 channels per timer. I am a newbie to rust and embedded programming so i am not very sure how this could be done using the pac crates but can anyone help regarding this?

strom-und-spiele, 2020-08-08:
Hey,

It helps a lot if you describe your setup in more detail (esp. what chip you are using), what your requirements are in detail and things you have tried so far. 

If you want to implement something using the PAC, check the reference manual of your chip for detailed information. 
E.g. for the F3Discovery, which is equipped with a STM32F303VC, [this RM ](https://www.st.com/resource/en/reference_manual/dm00043574-stm32f303xb-c-d-e-stm32f303x6-8-stm32f328x8-stm32f358xc-stm32f398xe-advanced-arm-based-mcus-stmicroelectronics.pdf) Has a detailed description of what to do in section 20.3.8.

 Of course generalization of your code to improve the current HAL are always welcome, even if it's just for your particular chip.

talalahsan01, 2020-08-08:
> Hey,
>
> It helps a lot if you describe your setup in more detail (esp. what chip you are using), what your requirements are in detail and things you have tried so far.
>
> If you want to implement [...]

Basically i am trying to use fs-ia6 receiver with my f3 board, the receiver outputs pwm signal ranging from 1000 to 2000 i need to receive this signal using tim2 of my board using 4 channels for example on receiver i have 6 channels i want to wire them up to pins pa0, pa1, pa2 and pa3 and i want to do use a function like "get_duty()" to calculate the duty cycle of incoming pwm signal from each pin. I hope that explains my setup

David-OConnor, 2020-09-15:
https://github.com/stm32-rs/stm32f3xx-hal/pull/141 includes the ability to set the `CaptureCompare` to input. Is still missing some features you might need though, but could be patched in easily. Have you figured out what register calls you need to make?

talalahsan01, 2020-10-26:
> #141 includes the ability to set the `CaptureCompare` to input mode. Is still missing some features you might need though, but could be patched in easily. Have you figured out what register calls you need to make?

Yup I have and i have made it work perfectly, but i still have issues in activating four channels at the same time, because the interrupt is global and I don't know how to set it up for each channel, nonetheless i still believe it would be a good idea to add it in the hal crate