stm32-rs/stm32f3xx-hal #134: Why eliminate USART3 for stm32f302 and stm32f334?

sprhawk, 2020-09-01:
https://github.com/stm32-rs/stm32f3xx-hal/blob/0ccda91eb9addfe7cb380c27b5d85eb844a5d384/src/serial.rs#L465

I checked the Programmaning manual, they both have usart3, why this is put separate condition?

strom-und-spiele, 2020-09-01:
I second this for the 334 variant (see [RM0364](https://www.st.com/resource/en/reference_manual/dm00093941-stm32f334xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf))
Also according to [RM0365](https://www.st.com/resource/en/reference_manual/dm00094349-stm32f302xb-c-d-e-and-stm32f302x6-8-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) there are some UART we could add on the 302 subvariants.

sprhawk, 2020-09-01:
@strom-und-spiele I'll send a PR soon

strom-und-spiele, 2020-09-01:
@sprhawk thanks, do you have those boards to verify the behavior?

sprhawk, 2020-09-01:
@strom-und-spiele I have one STM32F302R8, no F334.

sprhawk, 2020-09-01:
How about this to usart

```
 449 #[cfg(any(
 450     feature = "stm32f302",
 451     feature = "stm32f301",
 452     feature = "stm32f318",
 453     feature = "stm32f303",
 454     feature = "stm32f373",
 455     feature = "stm32f378",
 456     feature = "stm32f328",
 457     feature = "stm32f334",
 458     feature = "stm32f358",
 459     feature = "stm32f398"
 460 ))]
 461 hal! {
 462     USART1: (usart1, APB2, usart1en, usart1rst, pclk2),
 463     USART2: (usart2, APB1, usart2en, usart2rst, pclk1),
 464     USART3: (usart3, APB1, usart3en, usart3rst, pclk1),
 465 }
 466
 467 #[cfg(any(
 468     feature = "stm32f302xb",
 469     feature = "stm32f302xc",
 470     feature = "stm32f302xd",
 471     feature = "stm32f302xe",
 472     feature = "stm32f303xb",
 473     feature = "stm32f303xc",
 474     feature = "stm32f303xd",
 475     feature = "stm32f303xe",
 476 ))]
 477 hal! {
 478     USART4: (usart4, APB1, usart4en, usart4rst, pclk2),
 479 }
```

sprhawk, 2020-09-01:
#135

PR #139 (2020-09-11): Enable DMA for USART on STM32F302 devices
Replaces #135.
Closes #134.
closed 2020-09-11 by ra-kete