stm32-rs/stm32f3xx-hal #157: Thoughts on clock config

David-OConnor, 2020-10-25:
It seems like the current system fails silently, and gives the user too many degrees of freedom. Eg the illusion you can set whatever speeds you want. I think the STM32Cube IDE visual clock config is a good guide for what configs are valid.

Suggestion: Instead of setting speeds, set the various scalers. This alone reduces the degrees of freedom. Have a helper method to show what speeds these output. Calculate the speeds, and validate before setting. This deals with errors in 2 ways: Prevents requesting speeds that can't be set using the scalers and 2: Fails explicitly if the resulting speeds are out of range.

David-OConnor, 2020-10-25:
Closing in favor of https://github.com/stm32-rs/stm32f3xx-hal/pull/158

Sh3Rm4n, 2020-10-26:
Also some ideas came up in https://github.com/stm32-rs/stm32f3xx-hal/pull/156.

David-OConnor, 2021-01-05:
I just noticed that the [L0 hal implementation](https://github.com/stm32-rs/stm32l0xx-hal/blob/master/src/rcc.rs) is similar to the `clocks` module in my fork.

Sh3Rm4n, 2021-03-28:
This is an issue, I still want to tackle, but had no time as of now. I summarized my preferences how to implement a more robust and flexible clock configuration in https://github.com/stm32-rs/stm32f3xx-hal/pull/156#pullrequestreview-514497743