stm32-rs/stm32f3xx-hal #185: Leverage embedded-time

Sh3Rm4n, 2020-12-18:
We have some extension traits, which define times and frequencies in [`src/time.rs`](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/time.rs). 

This module is mostly a 1:1 copy, which can be found in almost every `stm32f*-hal`.
It mostly used internally, but gets also exposed, to configured clocks and timers. 

Wouldn't it be nicer to use `embedded-time` to replace the current [`time`](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/time.rs) module? 

Maybe we can leverage some more features of this crate.

wbuck, 2021-01-10:
So I forked the project, removed the custom `time` module and replaced it with [embedded-time](https://github.com/FluenTech/embedded-time/). It was almost a drop-in replacement. I altered the `serial` constructor to take a `Baud` type instead of the original `Bps`.

I only built it for the `stm32f303xc` but it did build with no issues.
I'm fairly new to `Rust` so I'm unsure what the expectation is regarding re-exporting a third party crate.
So to minimize changes I added the following to `lib.rs`

    pub use embedded_time as time;

David-OConnor, 2021-01-11:
Please make a PR.

strom-und-spiele, 2021-01-11:
@Sh3Rm4n thanks for proposing this. And thanks to @wbuck for giving a nice example on how to "leverage some more features of this crate" in  #191. 
Reading about its fixed-point arithmetic under the hood made me like this purpose even more :+1:

Sh3Rm4n, 2021-03-12:
Closed by https://github.com/stm32-rs/stm32f3xx-hal/pull/192

closed 2021-03-12 by Sh3Rm4n