stm32-rs/stm32f3xx-hal #190: Timers with period > 1s

David-OConnor, 2021-01-02:
Hi. Is it possible to implement timers with period > 1s? It seems that the `.hz()` api only takes integers.

If not, are you open to either a PR?

Sh3Rm4n, 2021-01-03:
I'm not sure, if it is really possible, but it should. 

Some pointers / ideas:

1. Hardware: Does the timer support frequencies lower than 1 Hz?
2. The [`Into<Hertz>`](https://github.com/stm32-rs/stm32f3xx-hal/blob/c68c36c03e0e33699b3b0c9acc3f8d80f5a25cd4/src/timer.rs#L141) has to be adjusted, as this currently limits the maximal timer period to 1 s.
   - Traits and `From<...>` conversions can be found [here](https://github.com/stm32-rs/stm32f3xx-hal/blob/c68c36c03e0e33699b3b0c9acc3f8d80f5a25cd4/src/time.rs). Maybe extending the [`U32Ext`](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/time.rs#L27-L43) implementation is useful? 
   - This is possible not enough, and this constraint has to be reworked to also support values in the `< 1 Hz` / `> 1 s` range 
3. (maybe) The [prescaler and auto-reload value calculation](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/timer.rs#L100-L109) has to be adjusted
    - Where `PSC` is the prescaler scaling down the peripheral clock which is connected to the timer block
    - And `ARR` is the auto-reload value, which holds the amount of timer ticks until an event / interrupt fires
    - The maximum value for `ARR` is the maximum possible length for a timer period in Hz, assuming `PSC` scales the peripheral clock down to 1 Hz

David-OConnor, 2021-01-03:
1. Yep! 

2. I think we'd need to bypass the `Into<Hertz>`, or fork it; I think it's the root cause. I remember looking at the source code a while back; I didn't understand how it worked. Reliance on it is the issue. I'd be happy setting period or frequency with a f32 argument, but that's a breaking change.

3. That's the core of it; PSC and ARR, in conjunction with clock speed are the only thing that determines period. For a given period, you set a combination of PSC and ARR. There is more than one combo for each period; we could pick arbitrarily.  I need to dive into the docs more, but IIIRC the period is (ARR+1) * (PSC+1) * the timer's clock speed.

Some naive approaches: Keep PSC and ARR equal. Calculate it with sqrt of period. (Has resolution issues). Fix one and set the other. (Has resolution and range issues). I think the existing code is pretty good about setting the right values, if they're faster than 1hz, and you leave it in edge mode.

Related: The latest version of my fork has a working interface to set RTC wakeup timer, where you pass period in second as a float. This is an easy-to-use interface that lets you select the full range; works similar to a timer, but the reg write process is different, and doesn't help here.

Sh3Rm4n, 2021-01-03:
> I think we'd need to bypass the Into<Hertz>, or fork it; I think it's the root cause

Definitely. This has to be reworked. I wouldn't worry about a breaking change too much. 
Maybe it would be worth to get feedback from the other stm hal impls as well, as they have to same issue / design IIRC. 

>  I'd be happy setting period or frequency with a f32 argument instead

Seems like a valid approach. I'm not sure though, if this does cause any issues. I wonder why the `u32` approach with `MegaHertz`, `Hertz`, `...` New-Type pattern is used everywhere. :thinking:

IamfromSpace, 2021-01-03:
Just to chime in, this is similar to the discussion around clocks for PWM.

Ultimately, the only way to get full control over these things is to basically give the same interface as the registers.  They use the divider -1 approach because all states are valid and you can get ever valid combo.  However, it’s just _terribly_ unfriendly to use.

It seems like there just needs to be multiple ways to configure these things, via trait or otherwise.  Possibly structs that can produce the register values given a clock frequency.

That way most users can get more conceptually useful inputs (like setting an approximate resulting period) or drill down to get exactly what they need.

David-OConnor, 2021-01-03:
I've added the method (as to be non-breaking) to my fork. Here's the code. The comments provide some details, including how this approach is quick+dirty: Ie there are ways to get better precision, perhaps at the expense of some computation.

```rust
  /// Set the timer period, in seconds. Overrides the period or frequency set
  /// in the constructor.
  /// This allows you to set periods greater than 1hz.
pub fn set_period(&mut self, period: f32, clocks: &clocks::Clocks) {
    // PSC and ARR range: 0 to 65535
    // (PSC+1)*(ARR+1) = TIMclk/Updatefrequency = TIMclk * period
    // APB1 (pclk1) is used by Tim2, 3, 4, 6, 7.
    // APB2 (pclk2) is used by Tim8, 15-20 etc.
    // todo: It appears there's a (fixed?) 2x multiplier on APB1
    // clocks; it's twice `pclk1`. See clocks diagram in RM, or `Clock Configuration`
    // tool in STM32CubeIDE.
    let tim_clk = clocks.calc_speeds().pclk1 * 1_000_000. * 2.;

    // We need to factor the right-hand-side of the above equation (`rhs` variable)
    // into integers. There are likely clever algorithms available to do this.
    // Some examples: https://cp-algorithms.com/algebra/factorization.html
    // We've chosen something quick to write, and with sloppy precision;
    // should be good enough for most cases.

    let rhs = tim_clk * period;

    // todo: Round instead of cast?
    let arr = (rhs.sqrt() - 1.) as u16;
    let psc = arr;

    self.tim.arr.write(|w| unsafe { w.bits(u32::from(arr)) });
    self.tim.psc.write(|w| unsafe { w.bits(u32::from(psc)) });
}
```

David-OConnor, 2021-01-04:
Probably should return a `Result` that fails if arr or psc is greater than 65k.

Sh3Rm4n, 2021-08-16:
Closed by #267

closed 2021-08-16 by Sh3Rm4n