stm32-rs/stm32f3xx-hal #191: Use embedded-time instead of custom implementation

wbuck, 2021-01-10:
When writing a driver for some device (say a motor driver) and attempt to program against the `trait`'s in the [embedded-hal](https://github.com/rust-embedded/embedded-hal) you can run into issues when using a `peripheral` that requires types defined in [time.rs](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/time.rs).

Because of this the driver now has a dependency on the `stm32f3xx-hal` device crate.
This of course is unwanted as it would be nice to create a `generic` device driver that doesn't depend on any specific device crate.

Example:

    pub struct Driver<T>
    where 
        T: CountDown
    { 
        timer: T,
    }

    impl<T> Driver<T> 
    where 
        T: CountDown
    {
        pub fn new( timer: T ) -> Self { Driver { timer }  }

        pub fn start( &self ) { 
             self.timer.start( /* This requires Hertz which is defined in time.rs of the stm32f3xx-hal crate */ );
        }
    }

Instead of using the units/functionality defined in [time.rs](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/time.rs) could the units/functionality defined in [embedded-time](https://github.com/FluenTech/embedded-time) be used instead? This would allow for a truly generic driver (assuming other device crates got on board).

This is probably more of community issue as a lot of device crates use their own custom `time.rs` implementation.

Below are two issues regarding this for the [embedded-hal](https://github.com/rust-embedded/embedded-hal) crate.
It sounded like the consensus was to have device crates get on board with using some centralized source for using time based units and functionality.

[embedded-hal #221](https://github.com/rust-embedded/embedded-hal/issues/211)
[embedded-hal #59](https://github.com/rust-embedded/embedded-hal/issues/59)

wbuck, 2021-01-10:
Just noticed that there was already a request for this:

[#191](https://github.com/stm32-rs/stm32f3xx-hal/issues/191)

Sh3Rm4n, 2021-03-12:
Closed by https://github.com/stm32-rs/stm32f3xx-hal/pull/192

closed 2021-03-12 by Sh3Rm4n