stm32-rs/stm32f3xx-hal #195: Low resolutions of pwm affect the frequency

seanybaggins, 2021-01-15:
using this code
```rust
use cortex_m_semihosting::{dbg, hprintln};
use panic_semihosting as _;

use stm32f3xx_hal as hal;
use hal::prelude::*;
use hal::pwm;

use cortex_m::asm;

use cortex_m_rt::entry;

#[cfg_attr(not(test), entry)]
fn main() -> ! {
    let device_peripherals = hal::pac::Peripherals::take().unwrap();
    // Setting the clock frequency to the maximum supported value
    let mut flash = device_peripherals.FLASH.constrain();
    let mut reset_and_control_clock = device_peripherals.RCC.constrain();
    let clocks = reset_and_control_clock
        .cfgr
        .sysclk(16.mhz())
        .freeze(&mut flash.acr);

    // Configuring PWM
    let resolution = 250;
    let frequency = 1.hz();
    let pwm_channel_no_pins =
        pwm::tim16(device_peripherals.TIM16, resolution, frequency, &clocks);
    let mut gpiob = device_peripherals.GPIOB.split(&mut reset_and_control_clock.ahb);
    let pb8 = gpiob.pb8.into_af1(&mut gpiob.moder, &mut gpiob.afrh);
    let mut pwm_channel = pwm_channel_no_pins
        .output_to_pb8(pb8);
    pwm_channel.set_duty(pwm_channel.get_max_duty() / 2);
    pwm_channel.enable();

    loop{}
```

I get a pwm signal that I expect. A 50% duty cycle with a 1 second period.

If I change `resolution = 200` then I get a signal that has a period of about 1/10 of the original.

seanybaggins, 2021-01-15:
Are there any tradeoffs to setting the resolution to u16::MAX. If not perhaps the solution does not need to be exposed to the user of the library.

IamfromSpace, 2021-01-16:
Hi @seanybaggins, original author of that particular module here.  The issue you're running into is a variant of the one mentioned in #138.  In that one, the resolution is set so high that the prescaler becomes too coarse to yield a reasonably close approximation to the desired frequency.  In your case, the resolution is set so low that prescaler overflows.

We have to set the prescaler so that: `timer_freq / prescaler == resolution * pwm_freq`.  Flipping that around to solve for the prescaler we get, `prescaler = timer_freq / resolution / pwm freq`.  In your first example (250), the result is 64000, which is _just_ inside our u16 maximum, giving us a valid prescaler value.  In the second example, we get 80000 which overflows.

I also link #138, because it goes into details about why this is more challenging to abstract away that I was originally hoping.  In fact, this interface was, I felt, the best hope for doing so, and you're not the first person to bump up against a leak.

As a tl;dr the issue is that you only have a limited number of bits, and must share them between the frequency accuracy and resolution fineness.  By putting them all in resolution, certain frequencies become essentially unachievable.

It's also challenging to be strict here.  This is configuration, so it's probably reasonable to `panic` if the result is obviously not going to make sense.  This may make sense in the case of prescaler overflow, but can't be done if the resulting frequency is approximate, because many frequencies can _only_ be approximately achieved!

I think the first thing to do is allow more flexibility in the ways that these values can be configured.  As an example, a user _should_ probably be able to set the values exactly if they so desire.  I've got a rough sketch of that I'll open as a PR for more feedback shortly.  Hopefully we can find something that moves the needle towards fewer gotchas :)

David-OConnor, 2021-01-17:
Bottom line: There are 3 things that affect frequency (or period): PSC register value, ARR register value, and timer clock speed. The current implementation is naive; it works well for some frequency and clock combos, but not others. Code from a fork I'm working on: It also is naive, and produces imprecise results at high frequencies. Check out the comments for a description of what's going on, and possible ways to address this.

```rust
/// Calculate values required to set the timer period: `PSC` and `ARR`. This can be
/// used for initial timer setup, or changing the value later.
fn calc_period_vals(period: f32, clocks: &clocks::Clocks) -> Result<(u16, u16), ValueError> {
    // PSC and ARR range: 0 to 65535
    // (PSC+1)*(ARR+1) = TIMclk/Updatefrequency = TIMclk * period
    // APB1 (pclk1) is used by Tim2, 3, 4, 6, 7.
    // APB2 (pclk2) is used by Tim8, 15-20 etc.
    // todo: It appears there's a (fixed?) 2x multiplier on APB1
    // timers; it's twice `pclk1`. See clocks diagram in RM, or `Clock Configuration`
    // tool in STM32CubeIDE.
    let tim_clk = clocks.calc_speeds().pclk1 * 1_000_000. * 2.;

    // We need to factor the right-hand-side of the above equation (`rhs` variable)
    // into integers. There are likely clever algorithms available to do this.
    // Some examples: https://cp-algorithms.com/algebra/factorization.html
    // We've chosen something quick to write, and with sloppy precision;
    // should be good enough for most cases.

    // - If you work with pure floats, there are an infinite number of solutions: Ie for any value of PSC, you can find an ARR to solve the equation.
    // - The actual values are integers that must be between 0 and 65_536
    // - Different combinations will result in different amounts of rounding errors. Ideally, we pick the one with the lowest rounding error.
    // - The aboveapproach sets PSC and ARR always equal to each other.
    // This results in concise code, is computationally easy, and doesn't limit
    // the maximum period. There will usually be solutions that have a smaller rounding error.

    let max_val = 65_535;
    let rhs = tim_clk * period;

    // todo: Round instead of cast?
    let arr = (rhs.sqrt() - 1.) as u16;
    let psc = arr;

    if arr > max_val || psc > max_val {
        return Err(ValueError {})
    }

    Ok((psc, arr))
}
```

David-OConnor, 2021-01-17:
I'm curious what the specific problem is here though: When I plug the numbers you posted into the `pwm` code above, the answer comes out right:

timer clock = 16,000,000. (Is this the problem? I'm not sure what the rcc module is doing with that code, when you specify `sysclk` only. Your timer clock in this case is APB2 (due to timer 16), possibly multiplied by 2 depending on how `rcc` is handling it)

arr = 200
freq = 1
psc = clock / arr / freq = 80,000

(80,000 + 1) * (200 + 1) / 16,000,000 = 1.005s period.

David-OConnor, 2021-01-17:
@seanybaggins: Could you please post the output of the following register reads, after the timer is set up?

```rust
unsafe { 
    (*pac::RCC::ptr()).cfgr2.read().prediv.bits();
    (*pac::RCC::ptr()).cfgr.read().pllmul.bits();
    (*pac::RCC::ptr()).cfgr.read().hpre.bits();
    (*pac::RCC::ptr()).cfgr.read().ppre2.bits();

    (*pac::TIM16::ptr()).psc.read().bits();
}
```

David-OConnor, 2021-01-17:
Here's one way to solve this:

Start with the square root approach I posted, where you set PSC and ARR equal to each other. Calculate speed. This will be close to the result. Add 1 to PSC and subtract 1 from ARR. Recalculate speed and compare to the initial result. If further from initial result, use initial result. If closer, repeat. When you find the closest point, use that.

This requires more computation upfront, so isn't suitable if you are repeatedly setting the period. It's still relatively cheap, and I think will get you the best result.

IamfromSpace, 2021-01-17:
Yeah, looking back at it, I was fairly sure that the resolution is off by one :/. The frequency is correct though I think.

And it does seem like there must be a simple/cheap alg that maximizes resolution against a target frequency.  Period freqs that are co-prime with the timer freq are troublesome in that the naive maximum resolution will be too small to be useful (whenever PSC is large enough, ARR will be 1).  In this case we need to somehow determine what the appropriate level of frequency accuracy is.

David-OConnor, 2021-01-17:
Could you clarify what problems being co-prime causes? I've been treating PSC and ARR as interchangeable, which may not be right. (Although it seems to work in practice, so far)

IamfromSpace, 2021-01-17:
Ah, yes, and treating them the same makes perfect sense in many contexts—because often only frequency matters.

PWM a place where that breaks down, because resolution is either somewhat or very important.  For something like a servo, you need a high enough resolution that you can express the positions.  If you range 180 across 5-10% duty cycle, then 3600 is going to allow you to express down to the degree.  If you need to hit _exact_ marks (say 10 degrees), then just 360 does the trick, but 577 misses them, even though it’s higher resolution.

Also, when we only care about frequency, it still makes sense to prioritize a larger PSC, as ARR is just soaking up bits, it doesn’t make it more accurate.

So, if we use PSC to maximize accuracy of a co-prime frequency we’ll get a very low arr:

timer_freq: 72_000_000
target_pwm_freq: 1_171
prescale_factor: 61_486
resolution: 1

We can never hit this frequency, so we use our entire prescaler value to get as close as possible.  That leaves no remaining clock divisions to share with ARR.

David-OConnor, 2021-01-17:
Thank you for the clarification. I've been using timers to fire interrupts and use 50%-duty PWM, so explains why I didn't notice a problem in my tests - and why the PWM module lets you specify ARR, but not the timer.

Lets say you're using that clock and target pwm freq. What trouble could you run into from setting `arr` and `psc` both to 247? This would yield a frequency of 1170.7Hz. Would the problem be in setting a small duty cycle, or triggering events midway through the duty cycle?

IamfromSpace, 2021-01-17:
Right, so 247 for 5-10% only gives you 12 useful values, and for 0-180 your jumps are about 15deg, which isn’t a lot of control.

And it just becomes hard to say what’s better, because more resolution creates more freq drift.

A 1171 frequency is certainly a worst case—I’m not sure there’s any reasonable value.  But it does a good job of illustrating the trade-off.

IamfromSpace, 2021-01-25:
Just edited my previous comment, realized that my previous statement wasn't quite correct: only the prescaler determines the accuracy of the frequency _when the resolution is fixed_.  If ARR has no constraints (like, be exactly x or be a multiple of y), then we get down to the original problem that you, @David-OConnor, had mentioned, how do we quickly and accurately factor our two dividers?  And instead of an even split, we likely want to have a lop-sided one, where resolution is prioritized.

Probably best to continue this discussion on #196, but I wanted to have this correction here :)