stm32-rs/stm32f3xx-hal #197: I2C doesn't work with repeating starts

David-OConnor, 2021-01-18:
This is part of the spec, and is required by some devices. Fixed [in this PR](https://github.com/stm32-rs/stm32f3xx-hal/pull/166). Copy+pasted `write_read` method from the H7 HAL.

eldruin, 2021-01-18:
Just weighing in here to point to the I2C specification corroborating this. See: [Figure 13 in section 3.1.10 of UM10204](https://www.nxp.com/docs/en/user-guide/UM10204.pdf#G1656228).

Piroro-hs, 2021-01-21:
I tested on stm32f303x8 and can't reproduce the issue...

David-OConnor, 2021-01-21:
What device did you test with? I2C works on many devices as-is, but the current non-compliant implementation fails on others. This brings it closer to I2C standards. (There's still a problem of Nacking at the end I haven't solved)

Piroro-hs, 2021-01-22:
According to [section 3.1.10](https://www.nxp.com/docs/en/user-guide/UM10204.pdf#G1656228), Master-receiver Nacking at the end is a spec compliant behavior.

closed 2021-03-28 by David-OConnor