stm32-rs/stm32f3xx-hal #205: [Bug] Incorrect pwm frequency

mstumbra, 2021-03-05:
`tim1(dp.TIM1, 144, 1.mhz().into(), &clocks)`
leads to actual timer frequency of 0.5 MHz, as code doesn't set/takes into account `tim1sw` bit in `RCC` `CFGR3` register
Same thing with other timers and possibly peripherals most likely. (Occurs only when MCU is running with max frequency)

David-OConnor, 2021-03-05:
Note that center align modes will also cause incorrect speeds.

Sh3Rm4n, 2021-03-05:
I have not had the time to dig into the PWM issues / PRs yet, but can this possible resolved or improved by https://github.com/stm32-rs/stm32f3xx-hal/pull/196?

Piroro-hs, 2021-03-14:
Looks like current PWM and Timer implementation is wrong at source clock selection.
https://github.com/stm32-rs/stm32f3xx-hal/blob/f8eb2926a38c9a40c41ba53de60839ac25d78f6e/src/pwm.rs#L283-L284

David-OConnor, 2021-03-14:
As in, some timers use apb1 while others use apb2?