stm32-rs/stm32f3xx-hal #214: impl trait for i2c

pdgilbert, 2021-04-10:
I have been struggling to discover the proper trait specification for `i2c` in the signature of  the `setup()` function in the following code.

<details>
<summary>Click to expand</summary>
use ssd1306::{prelude::*, Builder, I2CDIBuilder};

use stm32f3xx_hal::{
    gpio::{
        gpiob::{PB8, PB9},
        Alternate, OpenDrain, Pin, AF4,
    },
    //hal::blocking::i2c::{Read, Write, WriteRead},
    //hal::I2c,
    i2c::{I2c, SclPin, SdaPin},
    pac::{Peripherals, I2C1},
    prelude::*,
};

fn setup() -> I2c<I2C1, impl SclPin<I2C1> + SdaPin<I2C1>> {
    let p = Peripherals::take().unwrap();
    let mut rcc = p.RCC.constrain();
    let clocks = rcc.cfgr.freeze(&mut p.FLASH.constrain().acr);
    let mut gpiob = p.GPIOB.split(&mut rcc.ahb);

    let mut scl =
        gpiob
            .pb8
            .into_af4_open_drain(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrh); // scl on PB8
    let mut sda =
        gpiob
            .pb9
            .into_af4_open_drain(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrh); // sda on PB9

    scl.internal_pull_up(&mut gpiob.pupdr, true);
    sda.internal_pull_up(&mut gpiob.pupdr, true);

    // return i2c
    I2c::new(
        p.I2C1,
        (scl, sda),
        400_000.Hz(),
        clocks,
        &mut rcc.apb1,
    )
}

#[entry]
fn main() -> ! {
    let i2c = setup();

    let interface = I2CDIBuilder::new().init(i2c);
    let mut disp: GraphicsMode<_, _> = Builder::new()
        .size(DisplaySize128x64) // set display size 128x32, 128x64
        .connect(interface)
        .into();
    disp.init().unwrap();

    //builtin include Font6x6, Font6x8, Font6x12, Font8x16, Font12x16, Font24x32
    let text_style = TextStyleBuilder::new(Font8x16)
        .text_color(BinaryColor::On)
        .build();

    Text::new("Hello world!", Point::zero())
        .into_styled(text_style)
        .draw(&mut disp)
        .unwrap();

    Text::new("Hello Rust!", Point::new(0, 20))
        .into_styled(text_style)
        .draw(&mut disp)
        .unwrap();

    disp.flush().unwrap();

    loop {}
}

#[exception]
fn HardFault(ef: &ExceptionFrame) -> ! {
    panic!("{:#?}", ef);
}

</details>

The error from code above is
```
error[E0277]: the trait bound `(stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B0>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>, stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B1>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>): SclPin<I2C1>` is not satisfied
  --> examples/zzi2c.rs:30:25
   |
30 | fn setup() -> I2c<I2C1, impl SclPin<I2C1> + SdaPin<I2C1>> {
   |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `SclPin<I2C1>` is not implemented for `(stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B0>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>, stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B1>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>)`

error[E0277]: the trait bound `(stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B0>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>, stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B1>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>): SdaPin<I2C1>` is not satisfied
  --> examples/zzi2c.rs:30:25
   |
30 | fn setup() -> I2c<I2C1, impl SclPin<I2C1> + SdaPin<I2C1>> {
   |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `SdaPin<I2C1>` is not implemented for `(stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B0>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>, stm32f3xx_hal::gpio::Pin<Gpiob, typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, typenum::bit::B1>, Alternate<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B0>, OpenDrain>>)`

```

I have tried several variations on this. Previously I used the type specification, something like

```
fn setup() -> I2c<I2C1, (PB8<Alternate<AF4>>, PB9<Alternate<AF4>>)> 
```
but would like to use impl traits, especially since the type seems to have changed to something very long.

With other hals I can do `fn setup() -> I2c<I2C2, impl Pins<I2C2>> ` or `fn setup() -> I2c<I2C1, impl SclPin<I2C1>, impl SdaPin<I2C1>> ` but I have had no luck with `stm32f3xx_hal`.

Suggestions would be appreciated.

Piroro-hs, 2021-04-11:
Return type should be `I2c<I2C1, (impl SclPin<I2C1>, impl SdaPin<I2C1>)>`. (not tested)

pdgilbert, 2021-04-11:
Thanks, that it.

closed 2021-04-11 by pdgilbert