stm32-rs/stm32f3xx-hal #215: Getting it to run on STM32F303, can't compile

lucaszanella, 2021-04-19:
I'm very noob at this, would appreciate some help.

I'm trying the example https://github.com/stm32-rs/stm32f3xx-hal/blob/master/examples/pwm.rs on my STM32F303 Discovery board. Here's my cargo:

```
[package]
name = "stmf32f303_tests"
version = "0.1.0"
authors = ["Lucas Zanella <me@lucaszanella.com>"]
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
embedded-hal = "0.2.3"
nb = "0.1.2"
cortex-m = "0.6.2"
cortex-m-rt = "0.6.11"
# Panic behaviour, see https://crates.io/keywords/panic-impl for alternatives
panic-halt = "0.2.0"

[dependencies.stm32f3xx-hal]
version = "0.6.1"
features = ["rt", "stm32f303xc", "defmt"]
```

I've put the code of `pwm.rs` into `main.rs`. 

Errors:

```
error[E0599]: no method named `MHz` found for type `u32` in the current scope
  --> src/main.rs:31:40
   |
31 |     let clocks = rcc.cfgr.sysclk(16u32.MHz()).freeze(&mut flash.acr);
   |                                        ^^^ method not found in `u32`

error[E0599]: no method named `into_af2_push_pull` found for struct `stm32f3xx_hal::gpio::gpioa::PA4<Input<Floating>>` in the current scope
  --> src/main.rs:37:10
   |
37 |         .into_af2_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrl);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpioa::PA4<Input<Floating>>`

error[E0599]: no method named `into_af2_push_pull` found for struct `stm32f3xx_hal::gpio::gpioa::PA6<Input<Floating>>` in the current scope
  --> src/main.rs:40:10
   |
40 |         .into_af2_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrl);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpioa::PA6<Input<Floating>>`

error[E0599]: no method named `into_af2_push_pull` found for struct `stm32f3xx_hal::gpio::gpioa::PA7<Input<Floating>>` in the current scope
  --> src/main.rs:43:10
   |
43 |         .into_af2_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrl);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpioa::PA7<Input<Floating>>`

error[E0599]: no method named `into_af2_push_pull` found for struct `stm32f3xx_hal::gpio::gpiob::PB0<Input<Floating>>` in the current scope
  --> src/main.rs:48:10
   |
48 |         .into_af2_push_pull(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrl);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpiob::PB0<Input<Floating>>`

error[E0599]: no method named `into_af2_push_pull` found for struct `stm32f3xx_hal::gpio::gpiob::PB1<Input<Floating>>` in the current scope
  --> src/main.rs:51:10
   |
51 |         .into_af2_push_pull(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrl);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpiob::PB1<Input<Floating>>`

error[E0599]: no method named `into_af2_push_pull` found for struct `stm32f3xx_hal::gpio::gpiob::PB4<stm32f3xx_hal::gpio::AF0>` in the current scope
  --> src/main.rs:54:10
   |
54 |         .into_af2_push_pull(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrl);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpiob::PB4<stm32f3xx_hal::gpio::AF0>`

error[E0599]: no method named `into_af2_push_pull` found for struct `stm32f3xx_hal::gpio::gpiob::PB5<Input<Floating>>` in the current scope
  --> src/main.rs:57:10
   |
57 |         .into_af2_push_pull(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrl);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpiob::PB5<Input<Floating>>`

error[E0599]: no method named `into_af1_push_pull` found for struct `stm32f3xx_hal::gpio::gpiob::PB8<Input<Floating>>` in the current scope
  --> src/main.rs:60:10
   |
60 |         .into_af1_push_pull(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrh);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpiob::PB8<Input<Floating>>`

error[E0599]: no method named `into_af1_push_pull` found for struct `stm32f3xx_hal::gpio::gpiob::PB10<Input<Floating>>` in the current scope
  --> src/main.rs:63:10
   |
63 |         .into_af1_push_pull(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrh);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpiob::PB10<Input<Floating>>`

error[E0599]: no method named `into_af4_push_pull` found for struct `stm32f3xx_hal::gpio::gpioc::PC10<Input<Floating>>` in the current scope
  --> src/main.rs:68:10
   |
68 |         .into_af4_push_pull(&mut gpioc.moder, &mut gpioc.otyper, &mut gpioc.afrh);
   |          ^^^^^^^^^^^^^^^^^^ method not found in `stm32f3xx_hal::gpio::gpioc::PC10<Input<Floating>>`

error[E0599]: no method named `Hz` found for type `u32` in the current scope
  --> src/main.rs:76:15
   |
76 |         50u32.Hz(), // frequency of period
   |               ^^ method not found in `u32`

error[E0599]: no method named `Hz` found for type `u32` in the current scope
   --> src/main.rs:125:15
    |
125 |         50u32.Hz(), // frequency of period
    |               ^^ method not found in `u32`

error[E0599]: no method named `Hz` found for type `u32` in the current scope
   --> src/main.rs:140:15
    |
140 |         50u32.Hz(), // frequency of period
    |               ^^ method not found in `u32`

error[E0599]: no method named `Hz` found for type `u32` in the current scope
   --> src/main.rs:154:15
    |
154 |         50u32.Hz(), // frequency of period
    |               ^^ method not found in `u32`

```

Here's the entire project: https://github.com/lucaszanella/stmf32f303_tests_rust/tree/de0850645d8bca126fe6cb3f2cd04185d4d5631a

By the way, where's the blinky example for this board?

Rahix, 2021-04-19:
There are changes on the master branch which are not yet in the version from https://crates.io.  You should use the example from https://github.com/stm32-rs/stm32f3xx-hal/blob/v0.6.1/examples/pwm.rs probably.

lucaszanella, 2021-04-20:
@Rahix thanks. I also had to add `panic-semihosting = "0.5"` in cargo. Now I get

 ```
 = note: rust-lld: error: /workspaces/stmf32f303_tests/stmf32f03_tests/target/thumbv7m-none-eabi/debug/build/cortex-m-rt-ae57eb3f4f80966b/out/link.x:23: cannot find linker script memory.x
          >>> INCLUDE memory.x
          >>>         ^
          
```

even though memory.x is in the root of the project: https://github.com/lucaszanella/stmf32f303_tests_rust/tree/c539c1d96ee67aeae5b6360d967534f2371305e8

Where should I put it? I tried putting inside `.cargo` and `src` also, same error.

And do you have a blinky example for this board? I won't be able to test the PWM.

ps: I'll make a PR in this repo with better instructions once I make it work.

lucaszanella, 2021-04-20:
@Rahix 

I added memory.x to `/workspaces/stmf32f303_tests/stmf32f03_tests/target/thumbv7m-none-eabi/debug/build/cortex-m-rt-ae57eb3f4f80966b/out/` but I think this is not the right place. I get now:
```

root/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7m-none-eabi/lib/librustc_std_workspace_core-3a4e0255861739a3.rlib" "/root/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7m-none-eabi/lib/libcore-0d8d5cf15c7c9ebf.rlib" "--end-group" "/root/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7m-none-eabi/lib/libcompiler_builtins-534771455226d24e.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: undefined symbol: _defmt_acquire
          >>> referenced by export.rs:36 (/root/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.1.3/src/export.rs:36)
          >>>               defmt-84ed3a0cea1a91b2.defmt.6dx2hueh-cgu.4.rcgu.o:(defmt::export::acquire::h4db1f0ae6b2633e2) in archive /workspaces/stmf32f303_tests/stmf32f03_tests/target/thumbv7m-none-eabi/debug/deps/libdefmt-84ed3a0cea1a91b2.rlib
          
          rust-lld: error: undefined symbol: _defmt_release
          >>> referenced by export.rs:48 (/root/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.1.3/src/export.rs:48)
          >>>               defmt-84ed3a0cea1a91b2.defmt.6dx2hueh-cgu.4.rcgu.o:(defmt::export::release::hc6fca43b9f6a8bc3) in archive /workspaces/stmf32f303_tests/stmf32f03_tests/target/thumbv7m-none-eabi/debug/deps/libdefmt-84ed3a0cea1a91b2.rlib
          
          rust-lld: error: undefined symbol: _defmt_timestamp
          >>> referenced by export.rs:62 (/root/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.1.3/src/export.rs:62)
          >>>               defmt-84ed3a0cea1a91b2.defmt.6dx2hueh-cgu.4.rcgu.o:(defmt::export::timestamp::hc4235ffb1a53001c) in archive /workspaces/stmf32f303_tests/stmf32f03_tests/target/thumbv7m-none-eabi/debug/deps/libdefmt-84ed3a0cea1a91b2.rlib
          
          rust-lld: error: undefined symbol: _defmt_panic
          >>> referenced by export.rs:204 (/root/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.1.3/src/export.rs:204)
          >>>               defmt-84ed3a0cea1a91b2.defmt.6dx2hueh-cgu.4.rcgu.o:(defmt::export::panic::h6f1c2e6d454c15c3) in archive /workspaces/stmf32f303_tests/stmf32f03_tests/target/thumbv7m-none-eabi/debug/deps/libdefmt-84ed3a0cea1a91b2.rlib
          
```

do not know if it's related to where I've put it.

Sh3Rm4n, 2021-04-20:
Hey, as @Rahix correctly stated, the examples from the master branch are currently not building for the last released version. This should be fixed, when a new version is released, though it is advised to try out the examples under the corresponding version tag instead (in your case: [`examples/pwm.rs`](https://github.com/stm32-rs/stm32f3xx-hal/blob/v0.6.1/examples/pwm.rs))

The `memory.x` issue seems strange. Normally putting `memory.x` in the root folder of the project should work. For more detailed instructions, comparing against https://github.com/rust-embedded/cortex-m-quickstart is always helpful. 

Do you have a [`build.rs`](https://github.com/rust-embedded/cortex-m-quickstart/blob/master/build.rs) in your root project? This is needed, so that cargo can tell the linker (`rust-lld`) where to search for the linker-script (`memory.x`). 

Anyways, to the last problem you have: Does removing `defmt` from the features help?

```diff
  [dependencies.stm32f3xx-hal]
  version = "0.6.1"
- features = ["rt", "stm32f303xc", "defmt"]
+ features = ["rt", "stm32f303xc"]
```

lucaszanella, 2021-04-27:
@Sh3Rm4n removing fmt and putting memory.x in the right place and creating build.rs worked, it was one place off. Thank you very much. 

Could somebody link me with a blinking example? I'd like to see if things are working and I don't have any equipment to plug into the GPIO to test PWM. My discovery board has a build in LED (I guess).

lucaszanella, 2021-04-27:
I dockerized everything inside a docker container and also plugged it into VSCode's devcontainer so begginers can simply open this project in VSCode and it will just work, even the burning program will work.

https://github.com/lucaszanella/stmf32f303_tests_rust/tree/932a21dd6ac717b070f48634733ab627bcecb4ab

Just need to test the blinking example if someone can link me with one and see how it goes, then I'll probably write a nice introduction for newcomers.

Thanks for all the help.

Rahix, 2021-04-27:
What about [`toggle.rs`](https://github.com/stm32-rs/stm32f3xx-hal/blob/v0.6.1/examples/toggle.rs)?  You'll probably need to adjust which pin is used for the LED, of course.

lucaszanella, 2021-04-27:
@Rahix thanks, I was able to find the correct pin for one of the LEDs in my stm32f303. Are you guys using `cargo flash`? I tried and got

```
root@117f490caf45:/workspaces/stmf32f303_tests/stmf32f03_tests# cargo flash
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
    Flashing /workspaces/stmf32f303_tests/stmf32f03_tests/target/thumbv7m-none-eabi/debug/stmf32f303_tests
       Error The firmware on the probe is outdated
```

indeed this firmware is really old. Maybe some very recent firmware is needed for rust? Anyone got this problem before?

Sh3Rm4n, 2021-04-27:
Are you using the stm32f303-discovery board? It seems like the firmware for the onboard debugger is outdated and not supported by [cargo-flash](https://github.com/probe-rs/cargo-flash) / [probe-rs](https://github.com/probe-rs/probe-rs). [Here](https://github.com/probe-rs/probe-rs/issues/414#issuecomment-720415265) is an issue with similar problems, with some pointers how to update the debugger firmware. [This](https://www.st.com/en/development-tools/stsw-link007.html) could also help. 

I definitely did update the probe on my debugger some time ago, but do not remember the details to write down a detailed instruction. I hope these pointers will help anyways. :)

lucaszanella, 2021-04-27:
@Sh3Rm4n thanks, [this](https://my.st.com/cas/login?service=https%3A%2F%2Fmy.st.com%2Fcontent%2Fmy_st_com%2Fen%2Fproducts%2Fdevelopment-tools%2Fsoftware-development-tools%2Fstm32-software-development-tools%2Fstm32-programmers%2Fstsw-link007.license%3D1604289687555.product%3DSTSW-LINK007.version%3D2.37.26.html) worked and I was able to flash with `cargo flash --chip stm32f303C8 --release` after upgrading the firmware. Led blinks now. 

I just don't know if my `stm32f303` is the `C8` but it worked. Would be nice to know how to see this information in the board, so I can write into my little tutorial. Mine has nothing written into it, only `stm32f3`, I don't even remember how I know it's the `stm32f303`

Sh3Rm4n, 2021-04-28:
Great. Glad I could help you. As the original issue is resolved, I'll close this issue now. 

> I just don't know if my `stm32f303` is the `C8` but it worked. Would be nice to know how to see this information in the board, so I can write into my little tutorial. Mine has nothing written into it, only `stm32f3`, I don't even remember how I know it's the `stm32f303`

Well, if cargo flash did not complain, I guess it is not "too" wrong and memory mappings and checks inside the debugger succeeded. If you have the STM32F303 Discovery Board, it should be at least clear, that you have an stm32f303 as this should be printed on the PCB. 

Which variant of the stm32f303 you have can be either found out, if you look closely on the print of the MCU. Or you could look at the datasheet about your discovery board, or count the number of pins on your chip and match it too existing ones :grinning: 

My discovery board for example is the stm32f303 vct6 variant, therefore I would use `cargo flash --chip stm32f303vctx`.

Sh3Rm4n, 2021-04-28:
FYI, for `defmt` to work, you have to follow the [Application setup](https://defmt.ferrous-systems.com/setup-app.html). (I guess you missed the part, adjusting your `.cargo/config.toml` and this is the reason why `rust-ldd` has thrown errors)

Maybe the README / documentation should point to this guide to avoid confusion.

closed 2021-04-28 by Sh3Rm4n