stm32-rs/stm32f3xx-hal #218: cargo build --features stm32f303xc fails

cjheath, 2021-04-27:
Trying to build using current stable on OSX, I get four errors like this:

```
   Compiling stm32f3xx-hal v0.6.1 (/Users/cjh/work/electronics/ARM/Rust/stm32f3xx-hal)
error[E0425]: cannot find value `exti` in this scope
   --> src/gpio.rs:548:13
    |
548 |             $exti.[<$xr 1>]
    |             ^^^^^ not found in this scope
...
603 |             modify_at!(reg_for_cpu!(exti, rtsr), bitwidth, index, rise);
    |                        ------------------------ in this macro invocation
    |
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0425]: cannot find value `exti` in this scope
```
and three similar errors.

Sh3Rm4n, 2021-04-27:
> Trying to build using current stable (rust 1.45.2) on OSX, I get four errors like this:

Current stable is rust 1.51.0 and the CI currenlty builds with 1.48: 

https://github.com/stm32-rs/stm32f3xx-hal/blob/ff651241dc32b140709a2d3bcd05f9594b4f169a/.github/workflows/ci.yml#L62

Have you tried using a more recent compiler?

But this error message is kind of useless :/

cjheath, 2021-04-27:
Sorry, yes, that fixed it. I didn't think I was very far behind :)

closed 2021-04-27 by cjheath