stm32-rs/stm32f3xx-hal #228: USART value moved with split() method

adil-masood, 2021-05-26:
https://docs.rs/stm32f3xx-hal/0.6.1/stm32f3xx_hal/serial/struct.Serial.html
pub fn listen(&mut self, event: Event)
Starts listening for an interrupt event

pub fn unlisten(&mut self, event: Event)
Starts listening for an interrupt event

pub fn split(self) -> (Tx<USART1>, Rx<USART1>)
Splits the Serial abstraction into a transmitter and a receiver half

`       let mut usart1 = Serial::usart1(device.USART1, (pa9,pa10), 115_200.bps(), clocks,&mut rccx.apb2);
        let (mut tx, mut rx) = usart1.split();
        let mut listen = usart1.listen(serial::Event::Rxne);`

After using split() the usart1 is moved and i cant use listen() and unlisten()
I want to implement listen and unlisten with rtic.

Sh3Rm4n, 2021-05-26:
Yeah, this is currently a limitation of the current API. Only `Tx<USART1>` and `Rx<USART1>` are implementing the `send()` and `read()` functions.

What's needed to be done, is that [`Serial`](https://github.com/stm32-rs/stm32f3xx-hal/blob/dac205ea23502a8a038a438c4b8da2576958daa9/src/serial.rs#L113) has to implement the same methods as [`Rx`](vhttps://docs.rs/stm32f3xx-hal/0.6.1/stm32f3xx_hal/serial/struct.Rx.html) and [`Tx`](https://docs.rs/stm32f3xx-hal/0.6.1/stm32f3xx_hal/serial/struct.Tx.html). 

This is something I am aware of and is also a paint-point in this PR: [#212](https://github.com/stm32-rs/stm32f3xx-hal/pull/212/files#diff-c2ea97eb2a88d25f80387a578a8ced2fcd7551f315c215a01b6511d9fb28eabf)

Sh3Rm4n, 2021-06-08:
Hey. `Tx` and `Rx` still do not implement any `listen()` method, but as of #232 you shouldn't need to use `split()` to be able to read and write to and from the uart. This PR does even come with an example for rtic.

Does this cover your usecase?

That the USART module is moved into the Serial instance is desired behavior, and is not likly to change. To recover the value, you can use the `free()` method:

https://github.com/stm32-rs/stm32f3xx-hal/blob/2a4382e82c9df1470d58f1f3aefd98be38a5c9cb/src/serial.rs#L209

Sh3Rm4n, 2021-06-19:
Listen and unlisten is now implemented as well. #238 As well as the new version v0.7.0 was released, which includes both mentioned changes. 

I'll close this issue. If you still feel that the current implantation is missing anything for your described use case, feel free to comment here, so that I'll re-open this issue again.

closed 2021-06-22 by Sh3Rm4n