stm32-rs/stm32f3xx-hal #230: Provide release using peripheral access crate stm32f3 0.13.2

sirhcel, 2021-06-06:
There is an issue with `stm32f3` 0.13.0 and before which requires to use unsafe code for sending serial data via a USART (https://github.com/stm32-rs/stm32-rs/pull/558). This currently requires UART example code in the _Discovery Book_ to use `unsafe` (https://github.com/rust-embedded/discovery/pull/267#issuecomment-822826949) which does not play nice with an introduction to Embedded Rust.

Could you please release a version of this crate which uses `stm32f3` 0.13.2 (fixing the USART issue and https://github.com/stm32-rs/stm32-rs/issues/568) so that we can make the _Discovery Book_ examples safe again?

Sh3Rm4n, 2021-06-06:
Thanks for the heads up. This is a good motivation to actually release a new version. 

I'll plan to release a new version soon ish. I've took  the time of the weekend to bring #212  close to the finish line. This has to be cleaned up and merged. If this is done I'll plan to release a newer version.

sirhcel, 2021-06-18:
Release [0.7.0](https://github.com/stm32-rs/stm32f3xx-hal/releases/tag/v0.7.0) is out. Thank you very much @Sh3Rm4n! :)

closed 2021-06-18 by sirhcel