stm32-rs/stm32f3xx-hal #236: Re-export each pin type aliases (PA0, PA1, ..) under gpio module

Piroro-hs, 2021-06-08:
Pros:
* less typings
* remove redundancy in gpio::gpio**a**::P**A**0

Cons:
* Super long "Re-exports" section in rustdoc

Piroro-hs, 2021-06-08:
<details>
[image]
</details>

Looks not bad😃.
However, should we keep `gpio::gpioX::PXn` for compatibility?

Sh3Rm4n, 2021-06-08:
Hm. 🤔 I don't know? Maybe because this is subtle difference to the other implementation? But maybe not? It is definitely a big change for everyone upgrading a version. 

What about a slight conversation phase? Keep them in the old place for one version while introducing the new place and in the version after that remove them? 

After I'm finally about to release 0.7 I'll hope to increase the release cadence.

closed 2021-07-23 by Piroro-hs