stm32-rs/stm32f3xx-hal #248: Can't use SPI in 0.7.0 release

hallgchris, 2021-06-26:
Hi there, pretty new to embedded rust so sorry if this isn't a good place to ask for help as I imagine this is on me, not a bug. Would love to be directed to the right place :)

Using the 0.6.1 release of this library I was using SPI to drive some WS2812b LEDs, creating the SPI instance like so:
```rust
    // Set up SPI for WS2812b LEDs
    let (sck, miso, mosi) = (
        gpioa.pa5.into_af5(&mut gpioa.moder, &mut gpioa.afrl),
        gpioa.pa6.into_af5(&mut gpioa.moder, &mut gpioa.afrl),
        gpiob.pb5.into_af5(&mut gpiob.moder, &mut gpiob.afrl),
    );
    let spi = Spi::spi1(
        dp.SPI1,
        (sck, miso, mosi),
        ws2812_spi::MODE,
        3.mhz(),
        clocks,
        &mut rcc.apb2,
    );
    let mut board_leds = Ws2812::new(spi);
```

After updating to 0.7.0, my code looks very similar:
```rust
    // Set up SPI for WS2812b LEDs
    let (sck, miso, mosi) = (
        gpioa
            .pa5
            .into_af15_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrl),
        gpioa
            .pa6
            .into_af5_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrl),
        gpiob
            .pb5
            .into_af5_push_pull(&mut gpiob.moder, &mut gpiob.otyper, &mut gpiob.afrl),
    );
    let mut spi = Spi::spi1(
        dp.SPI1,
        (sck, miso, mosi),
        ws2812_spi::MODE,
        3.MHz().try_into().unwrap(),
        clocks,
        &mut rcc.apb2,
    );
    let mut board_leds = Ws2812::new(spi);
```

However, it doesn't compile; I get the following error:
```
error[E0277]: the trait bound `stm32f3xx_hal::gpio::Pin<Gpioa, U<5_u8>, Alternate<PushPull, 15_u8>>: SckPin<stm32f3xx_hal::pac::SPI1>` is not satisfied
  --> firmware/src/main.rs:49:19
   |
49 |     let mut spi = Spi::spi1(
   |                   ^^^^^^^^^ the trait `SckPin<stm32f3xx_hal::pac::SPI1>` is not implemented for `stm32f3xx_hal::gpio::Pin<Gpioa, U<5_u8>, Alternate<PushPull, 15_u8>>`
   |
   = help: the following implementations were found:
             <stm32f3xx_hal::gpio::Pin<Gpioa, U<5_u8>, Alternate<PushPull, 5_u8>> as SckPin<stm32f3xx_hal::pac::SPI1>>
             <stm32f3xx_hal::gpio::Pin<Gpiob, U<13_u8>, Alternate<PushPull, 5_u8>> as SckPin<stm32f3xx_hal::pac::SPI2>>
             <stm32f3xx_hal::gpio::Pin<Gpiob, U<3_u8>, Alternate<PushPull, 5_u8>> as SckPin<stm32f3xx_hal::pac::SPI1>>
             <stm32f3xx_hal::gpio::Pin<Gpioc, U<10_u8>, Alternate<PushPull, 6_u8>> as SckPin<stm32f3xx_hal::pac::SPI3>>
   = note: required by `Spi::<stm32f3xx_hal::pac::SPI1, (SCK, MISO, MOSI), WORD>::spi1`
```

I feel like it's got to be something silly I've missed, but I'm pretty stumped as to how it used to work but now it doesn't. (Maybe I just need to get some sleep...) I've compared it to the `spi.rs` example in this repo and I don't see any real difference.

I'm programming for an stm32f303x8.

Any ideas why this doesn't work or a referral to a better place to ask this would be hugely appreciated! Thanks for your time, this crate is awesome and I'm super excited to use 0.7.0 - there's quite a few changes I've had my eye on :)

adamgreig, 2021-06-26:
In this line of your new code:

```rust
            .into_af15_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrl),
```

Is that a typo and it should be `af5` instead?

hallgchris, 2021-06-26:
Oh dear, that's almost certainly it. Can't believe I couldn't see it! Thanks heaps :)

closed 2021-06-26 by hallgchris