stm32-rs/stm32f3xx-hal #268: Use of build-dependency cargo-metadata breaks no_std crates that use serde

ho-ho-ho, 2021-08-05:
`cargo-metadata` depends on `serde` with std enabled.
Using another crate which depends on `serde` - even when used in a no_std context - therefore forces usage of std (related to https://github.com/rust-lang/cargo/issues/5730 ), rendering it unable to compile for a thumb target.
I've ran into this issue today when trying to use the `usbd-hid` crate and it failed to compile because of that :(

Would it be possible to remove the dependency on `cargo-metadata`?

ho-ho-ho, 2021-08-05:
Well looks like I posted too soon ... 
for everyone running into this problem: the new cargo resolver already fixed that (adding `resolver = "2"` to Cargo.toml package section works).

closed 2021-08-05 by ho-ho-ho

Sh3Rm4n, 2021-08-06:
Thanks for reporting that either way. 

It is definitely important to know, that it does not work with `resolver = "1"` and as that is still the default IIRC it should be mentioned somewhere to use the newer resolver. 

Fortunately the MSRV is 1.51 because of minimal const generics support and that is also the version, where the new resolver is released.