stm32-rs/stm32f3xx-hal #288: i2c type alias

pdgilbert, 2021-10-04:
(Related to #214) I am trying to define a type alias for an i2c interface because I need it in `rtic` `struct Shared`. I've tried several variations with no success. For example, `type I2cBus = I2c<I2C1, (PB6<Alternate<OpenDrain, AF4>>, PB7<Alternate<OpenDrain, AF4>>)>` gives
```
  --> examples/zzot.rs:44:9
   |
44 |         pac,
   |         ^^^
...
47 |         serial::{RxPin, Serial, Tx, TxPin},
   |                  ^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

error[E0747]: type provided when a constant was expected
  --> examples/zzot.rs:57:55
   |
57 |     type I2cBus = I2c<I2C1, (PB6<Alternate<OpenDrain, AF4>>, PB7<Alternate<OpenDrain, AF4>>)>;
   |                                                       ^^^
   |
help: if this generic argument was intended as a const parameter, surround it with braces
   |
57 |     type I2cBus = I2c<I2C1, (PB6<Alternate<OpenDrain, { AF4 }>>, PB7<Alternate<OpenDrain, AF4>>)>;
   |                                                       ^     ^

error[E0747]: type provided when a constant was expected
  --> examples/zzot.rs:57:87
   |
57 |     type I2cBus = I2c<I2C1, (PB6<Alternate<OpenDrain, AF4>>, PB7<Alternate<OpenDrain, AF4>>)>;
   |                                                                                       ^^^
   |
help: if this generic argument was intended as a const parameter, surround it with braces
   |
57 |     type I2cBus = I2c<I2C1, (PB6<Alternate<OpenDrain, AF4>>, PB7<Alternate<OpenDrain, { AF4 }>>)>;
   |                                                                                       ^     ^

error: aborting due to 2 previous errors; 1 warning emitted
```
but surrounding in brackets makes things worse.  (I'm using git `stm32f3xx-hal#53567500d7ab0`. With different `setup()` functions this example works with several other device hals.)  Suggestions would be appreciated. 

The complete example code is 
<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;
//use panic_rtt_target as _;

use rtic::app;

#[cfg_attr(feature = "stm32f3xx", app(device = stm32f3xx_hal::pac,   dispatchers = [TIM2, TIM3]))] 

mod app {

    use dwt_systick_monotonic::DwtSystick;
    use rtic::time::duration::{Seconds, };

    use cortex_m::asm; //asm::delay(N:u32) blocks the program for at least N CPU cycles.

    use core::fmt::Write;
    use embedded_ccs811::{
        mode as Ccs811Mode, prelude::*, AlgorithmResult, Ccs811Awake, MeasurementMode,
        SlaveAddr as Ccs811SlaveAddr,
    };
[... 200 lines cut ...]
            let en = cx.shared.hdc2080.lock(|hdc2080| block!(hdc2080.read())).unwrap();
            let temp = en.temperature;
            let humidity = en.humidity.unwrap_or(0.0);
            cx.local.env[*cx.local.index] = (temp, humidity);
            *cx.local.index += 1;
            //cx.shared.ccs811.set_environment(temp, humidity).unwrap();
            cx.shared.ccs811.lock(|ccs811| ccs811.set_environment(temp, humidity)).unwrap();
        }
        //writeln!(cx.shared.tx, "\rstart\r",).unwrap();
        cx.shared.tx.lock(|tx| writeln!( tx, "\rstart\r",)).unwrap();
        for i in 0..*cx.local.index {
            let data = cx.local.measurements[i];
            let en = if i == 0 { (0.0, 0.0) } else { cx.local.env[i - 1] };
            //writeln!(cx.shared.tx,  "{},{},{},{},{},{:.2},{:.2}\r", i, data.eco2, 
            //        data.etvoc, data.raw_current, data.raw_voltage, en.0, en.1 ).unwrap();
            cx.shared.tx.lock(|tx| writeln!(tx,  "{},{},{},{},{},{:.2},{:.2}\r", i, data.eco2, 
                    data.etvoc, data.raw_current, data.raw_voltage, en.0, en.1 )).unwrap();
        }
        //cx.schedule.measure(cx.scheduled + PERIOD.cycles()).unwrap();
        measure::spawn_after(PERIOD).unwrap();
    }

}
```
</details>

Sh3Rm4n, 2021-10-04:
Without checking, either 

```rust

type I2cBus = I2c<I2C1, (PB6<Alternate<OpenDrain, { 4_u8 }>>, PB7<Alternate<OpenDrain, { 4_u8 }>>)>;
```

or 

```rust
type I2cBus = I2c<I2C1, (PB6<AF4<OpenDrain>>, PB7<AF4<OpenDrain>>)>;
```

should work.

[`AF4`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/gpio/type.AF4.html) in itself is a type alias, which hides the fact, that const generics are used to define  the type.

I know the compiler gives some misleading (or not easy to understand) suggestions on these highly generic types, so only relying on the compiler sadly leads to the trial and error you stepped into (I have myself in the past, so this part needs to definitely be improved, most probably on stm32f3xx-hal's side) 

As you can see [`SclPin`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/i2c/trait.SclPin.html) for example has some [implementors](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/i2c/trait.SclPin.html#implementors) 

[image]

So choosing one of these for the type alias should give you the quickest success.

pdgilbert, 2021-10-04:
Thanks @Sh3Rm4n, both of those work. Also thanks for the pointers on finding type definitions. I've been down the rabbit hole of following compiler suggestions several times. I'm now having a slightly different problem with `doesn't have a size known at compile-time` for the TxType.  Adding `dyn` or not just removes a warning.  Any suggestins would be appreciated. I'm getting

```
Compiling rust-integration-testing-of-examples v0.1.0 (/home/paul/githubClones/rust-integration-testing/dev-testing)
error[E0277]: the size for values of type `(dyn stm32f3xx_hal::serial::TxPin<USART1> + 'static)` cannot be known at compilation time
   --> examples/zzot.rs:63:34
    |
63  |     fn setup(dp: Peripherals) -> (I2cBus, LedType, TxType ) {
    |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^ doesn't have a size known at compile-time
    | 
   ::: /home/paul/.cargo/git/checkouts/stm32f3xx-hal-572339a820137a38/5356750/src/serial.rs:346:26
    |
346 |     pub struct Tx<Usart, Pin> {
    |                          --- required by this bound in `stm32f3xx_hal::serial::Tx`
    |
    = help: the trait `Sized` is not implemented for `(dyn stm32f3xx_hal::serial::TxPin<USART1> + 'static)`

error[E0277]: the size for values of type `(dyn stm32f3xx_hal::serial::TxPin<USART1> + 'static)` cannot be known at compilation time
   --> examples/zzot.rs:137:14
    |
137 |         tx : TxType,
    |              ^^^^^^ doesn't have a size known at compile-time
    | 
   ::: /home/paul/.cargo/git/checkouts/stm32f3xx-hal-572339a820137a38/5356750/src/serial.rs:346:26
    |
346 |     pub struct Tx<Usart, Pin> {
    |                          --- required by this bound in `stm32f3xx_hal::serial::Tx`
    |
    = help: the trait `Sized` is not implemented for `(dyn stm32f3xx_hal::serial::TxPin<USART1> + 'static)`

error[E0277]: the size for values of type `(dyn stm32f3xx_hal::serial::TxPin<USART1> + 'static)` cannot be known at compilation time
   --> examples/zzot.rs:14:35
    |
14  | #[cfg_attr(feature = "stm32f3xx", app(device = stm32f3xx_hal::pac,   dispatchers = [TIM2, TIM3]))] 
    |                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ doesn't have a size known at compile-time
    | 
   ::: /home/paul/.cargo/git/checkouts/stm32f3xx-hal-572339a820137a38/5356750/src/serial.rs:346:26
    |
346 |     pub struct Tx<Usart, Pin> {
    |                          --- required by this bound in `stm32f3xx_hal::serial::Tx`
    |
    = help: the trait `Sized` is not implemented for `(dyn stm32f3xx_hal::serial::TxPin<USART1> + 'static)`
    = note: this error originates in an attribute macro (in Nightly builds, run with -Z macro-backtrace for more info)

error: aborting due to 3 previous errors

```

Sh3Rm4n, 2021-10-05:
You are using 

```rust
type TxType = Tx<USART1, dyn TxPin<USART1>>;
```

which internally already uses `dyn`, which is `!Sized` (unsized) by default (IIRC) so you either have to `Box` it, which is not possible in `no_std` or use `&dyn TxPin`. **However**, I think you do not need to reach for dynamic dispatch here, what you are trying to do, is to create a type alias for a **concrete** type you already know at this point. 

Looking at the definition of [`Tx`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/serial/struct.Tx.html#impl) you have to provide a [`serial::Instance`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/serial/trait.Instance.html), which you have done already ([`USART1`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/pac/struct.USART1.html)) **and** have to provide a pin, which implements [`TxPin`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/serial/trait.TxPin.html). 

Looking at your code, you are trying to use PA9: 

```rust
        let (tx, _rx) = Serial::usart1(
            dp.USART1,
            (
                gpioa
                    .pa9
                    .into_af7_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrh), //tx pa9
                gpioa
                    .pa10
                    .into_af7_push_pull(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrh), //rx pa10
            ),
            115200.Bd(),
            clocks,
            &mut rcc.apb2,
        )
        .split();
```

which is implementing `TxPin`: [`impl<Otype> TxPin<USART1> for PA9<AF7<Otype>>`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/serial/trait.TxPin.html#impl-TxPin%3CUSART1%3E). 

So the complete type is: 

```rust
type TxType = Tx<USART1, PA9<AF7<OpenDrain>>>;
```

pdgilbert, 2021-10-05:
Thank you again @Sh3Rm4n. Indeed I was not trying to do dynamic dispatch and only got there by following compiler suggestions. The `type TxType = Tx<USART1, PA9<AF7<OpenDrain>>>;` works but I am curious why I do not have to specify the pin with other hals. These all work:
```
    #[cfg(feature = "stm32f1xx")]
    type TxType = Tx<USART1>;

    #[cfg(feature = "stm32f4xx")]
    type TxType = Tx<USART1>;

    #[cfg(feature = "stm32f7xx")]
    type TxType = Tx<pac::USART2>;

   #[cfg(feature = "stm32h7xx")]
    type TxType = Tx<USART2>;

    #[cfg(feature = "stm32l1xx")]
    type TxType = Tx<USART1>;

    #[cfg(feature = "stm32l4xx")]
    type TxType = Tx<USART2>;
```

Sh3Rm4n, 2021-10-05:
This is because in other HALs, the [`split()`](https://docs.rs/stm32f4xx-hal/0.10.0/stm32f4xx_hal/serial/struct.Serial.html#method.split) does return a `Tx` type, which looses the information, from what pin it was created. 

For the `stm32f3xx-hal`, this information is kept: See [`Serial::split()`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/serial/struct.Serial.html#method.split).

This was necessary it to be able to implement [`Serial::join()`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/serial/struct.Serial.html#method.join) implemented in #252. This functionality is not provided by the other HALs.

pdgilbert, 2021-10-05:
That makes sense. At this rate I might actual start to understand these interfaces someday. Thanks for the explanation.

closed 2021-10-05 by pdgilbert