stm32-rs/stm32f3xx-hal #303: Pin usage clarity for peripheral (special case for PB3/PB4 for SPI3)

mkdxdx, 2021-12-12:
I've just started using this crate for stm32f303vc on discovery board and immediately ran into this issue that i'm not able to use PB3 and PB4 as pins for SPI3 peripheral.
```    
    let mut gpiob = periph.GPIOB.split(&mut rcc.ahb);
    let mut gpioc = periph.GPIOC.split(&mut rcc.ahb);
    let pb3_sck = gpiob.pb3.into_af6_push_pull(
        &mut gpiob.moder, 
        &mut gpiob.otyper, 
        &mut gpiob.afrl);
    let mut pb4_miso = gpiob.pb4.into_af6_push_pull(
        &mut gpiob.moder, 
        &mut gpiob.otyper, 
        &mut gpiob.afrl);
    let mut pc12_mosi = gpioc.pc12.into_af6_push_pull(
        &mut gpioc.moder, 
        &mut gpioc.otyper, 
        &mut gpioc.afrh);
    let mut spi3 = Spi::new(
        periph.SPI3, 
        (pb3_sck, pb4_miso, pc12_mosi), 
        8.MHz(),
        clocks,
        &mut rcc.apb1);
```
which gives me
```
the trait SckPin<stm32f3xx_hal::pac::SPI3> is not implemented for stm32f3xx_hal::gpio::Pin<Gpiob, U<3_u8>, Alternate<PushPull, 6_u8>>

the trait MisoPin<stm32f3xx_hal::pac::SPI3> is not implemented for stm32f3xx_hal::gpio::Pin<Gpiob, U<4_u8>, Alternate<PushPull, 6_u8>>
```
Now, according to datasheet, these pins _can_ be technically used for SPI3 as SCK and MISO in AF6 mode, however by default and after reset they function as debug pins for JTAG/SWD, and this, i only assume, may be why these pins have no trait implementation.
However i found no explicit reasoning as to why these pins simply cannot be used with a peripheral. And implementing my own trait SckPin/MisoPin doesn't work since they are either sealed, or, as in alternate stm32f30x HAL repository, [explicitly asked not to be implemented](https://github.com/japaric/stm32f30x-hal/blob/3433e5463478c0dab8a2afea5e27cf815af89720/src/spi.rs#L30).

Sorry if i'm missing a portion of discussion somewhere in this repo where this approach was settled and used.

A compile time assertion fault with explicit error message could drastically help with that.

adamgreig, 2021-12-12:
For the F302 and F303xD/E those traits are implemented [here](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/spi.rs#L143-L151) behind the `gpio-f302` and `gpio-f303e` [features](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/Cargo.toml#L115-L117), but not for the F303xC you're using. The datasheet does indeed show they can be used, so this is probably just a bug/omission rather than a deliberate choice to restrict the use of the SWD pins. Without checking the datasheets in more detail I'm not sure of the other differences between `gpio-303` and `gpio-f303e` but maybe Sck and Miso can be implemented for PB3/PB4 on `gpio-f303`.

Sh3Rm4n, 2021-12-12:
Re-checking the datasheets for all `gpio-f303` devices (exemplary `stm32f303vc`, `stm32f302vc` and `stm32f358vc`) there is no reason, that SCK and MISO shouldn't be implemented. This is definitely an oversight. Thanks for the catch! :)

Sh3Rm4n, 2021-12-12:
Hey @mkdxdx, can you verify #304 is fixing your issue. Easiest way is to add `[patch]` to your `Cargo.toml` and run `cargo update` afterwards:

```toml
[patch.crates-io]
stm32f3xx-hal = { git = 'https://github.com/Sh3Rm4n/stm32f3xx-hal', branch = 'fix-spi-impls' }
```

mkdxdx, 2021-12-12:
@Sh3Rm4n thanks for quick reply!

I've tried by your instruction add patch and now it gives different instantiation error (with same code)
```
  --> src\main.rs:67:19
   |
67 |     let mut spi = Spi::new(
   |         -------   ^^^^^^^^ cannot infer type for type parameter `WORD`
   |         |
   |         consider giving `spi` the explicit type `Spi<stm32f3xx_hal::pac::SPI3, (stm32f3xx_hal::gpio::Pin<Gpiob, U<3_u8>, Alternate<PushPull, 6_u8>>, stm32f3xx_hal::gpio::Pin<Gpiob, U<4_u8>, Alternate<PushPull, 6_u8>>, stm32f3xx_hal::gpio::Pin<Gpioc, U<12_u8>, Alternate<PushPull, 6_u8>>), Word>`, where the type parameter `WORD` is specified
```

mkdxdx, 2021-12-12:
Got it working by looking up correct usage in [test suite](https://github.com/stm32-rs/stm32f3xx-hal/blob/v0.8.1/testsuite/tests/spi.rs). I guess example code should also be updated. 

Since my code is able to build, i guess it's resolved then, thanks!

Sh3Rm4n, 2021-12-12:
Oh yeah, the `WORD` specifier. Not really obvious. I guess you are not invoking any `send()` or `read()` method in your example code, with which the compiler could infer the correct type for WORD. 

I guess updating the example, so it specifies word, does not hurt. Thanks for pointing that out. 

I'll try to finish #304 tomorrow and maybe also try to release `v0.8.2` earlier next week :)

closed 2021-12-13 by Sh3Rm4n

Sh3Rm4n, 2021-12-14:
The fix should be release with [v0.8.2](https://crates.io/crates/stm32f3xx-hal/0.8.2) now :)

PR #304 (2021-12-13): Add missing spi impls for gpio-f303
- [ ] Adjust example to specify [`WORD`](https://github.com/stm32-rs/stm32f3xx-hal/issues/303#issuecomment-991955400)
- [ ] Fix testsuite build

Fixes #303
commit 54770755 (2021-12-13): Add missing spi impls for gpio-f303

Fixes #303
commit 1cdb1989 (2021-12-13): Add missing spi impls for gpio-f303

Fixes #303