stm32-rs/stm32f3xx-hal #324: examples/pwm.rs Outdated/depreciated?

tomasApo, 2022-11-23:
### STM32f3 Discovery board to work with PWM. 

Unfortunately, I keep having to make alterations to the example code to try satisfy the compiler. Is it possible that the [examples/pwm.rs ](https://github.com/stm32-rs/stm32f3xx-hal/blob/c68c36c03e0e33699b3b0c9acc3f8d80f5a25cd4/examples/pwm.rs) is outdated to the point it no longer runs with recent versions of stm32f3xx-hal?

My suspicion comes from conflicting examples, as a beginner I struggle to read the doc.rs, so I find the example sections very useful. Unfortunately, https://docs.rs/stm32f3xx-hal/latest/stm32f3xx_hal/index.html doesn't match with  [examples/pwm.rs ](https://github.com/stm32-rs/stm32f3xx-hal/blob/c68c36c03e0e33699b3b0c9acc3f8d80f5a25cd4/examples/pwm.rs) 

Should I downgrade my `stm32f3xx-hal = version = "0.9.1";` to `0.6.1`?

Sh3Rm4n, 2022-11-23:
Hey @tomasApo . I'm glad you are using this hal, really appreciate it. Unfortunatly I haven't had the time to maintain this project for a while. Refactoring the pwm module is on my TODO list for quite a while as well, as it does not really fit the rest of the coding style and it is difficult to use in more generic code. 

Anyways, I haven't used the pwm implementation myself except running the example sporadically to see if it does not panic. But I haven't checked if it is actually working. Also tests are missing for the PWM module. 

So I guess your best bett is to downgrade to "0.6.1" for now and see if it works. If this fixes the issue, it must be a regression. But I haven't don't any major changes to the PWM module since, so I assume it is also not that hard to fix a potential regression.

tomasApo, 2022-11-23:
No worries, I find the documentation of your crate far easier. Specially with the examples.

Here is a simple example working with the `0.6.1` https://github.com/tomasApo/stm32f3_pwm_v0

https://user-images.githubusercontent.com/75183079/203642405-478770e8-798f-4141-9663-9e7a0b7543ac.MOV

😁

closed 2022-12-22 by tomasApo