stm32-rs/stm32f3xx-hal #332: Issue with ADC channel/pin mapping

jack-greenberg, 2023-02-01:
I'm not 100% sure if I'm just implementing incorrectly, but I am unable to use one of the ADC channels for a pin that I believe it is supposed to work for.

Chip: STM32F303RBTx
ADC: ADC4
Channel: 5
Pin: 15

Here's the code:

```rust
    // Setup clocks and such...

    // Gpio setup
    let mut gpiob = dp.GPIOB.split(&mut rcc.ahb);

    let adc_34_common = adc::CommonAdc::new(dp.ADC3_4, &clocks, &mut rcc.ahb);

    let mut _adc3 = adc::Adc::new(
        dp.ADC3, // The ADC we are going to control
        adc::config::Config::default(),
        &clocks,
        &adc_34_common,
    ).into_oneshot();

    let mut adc4 = adc::Adc::new(
        dp.ADC4, // The ADC we are going to control
        adc::config::Config::default(),
        &clocks,
        &adc_34_common,
    ).into_oneshot();

    let mut test_pin = gpiob.pb15.into_analog(&mut gpiob.moder, &mut gpiob.pupdr);

    loop {
        let test_value: u16 = adc4.read(&mut test_pin).unwrap();

        defmt::trace!("{}", test_value as f32 * 3.3 / 4096.0);
    }
```

<details>
<summary>And the output:</summary>

```
error[E0277]: the trait bound `stm32f3xx_hal::gpio::Pin<Gpiob, U<15>, Analog>: stm32f3xx_hal::embedded_hal::adc::Channel<stm32f3xx_hal::pac::ADC4>` is not satisfied
  --> roadkill/src/main.rs:71:46
   |
71 |         let pressure_before: u16 = adc4.read(&mut pressure_before_pin).unwrap();
   |                                         ---- ^^^^^^^^^^^^^^^^^^^^^^^^ the trait `stm32f3xx_hal::embedded_hal::adc::Channel<stm32f3xx_hal::pac::ADC4>` is not implemented for `stm32f3xx_hal::gpio::Pin<Gpiob, U<15>, Analog>`
   |                                         |
   |                                         required by a bound introduced by this call
   |
   = help: the following other types implement trait `stm32f3xx_hal::embedded_hal::adc::Channel<ADC>`:
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<0>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC1>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<1>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC1>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<3>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC1>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<4>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<5>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<6>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<7>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpiob, U<0>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<stm32f3xx_hal::pac::ADC3>>
           and 34 others
note: required by a bound in `stm32f3xx_hal::prelude::_embedded_hal_adc_OneShot::read`
  --> /home/jack/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-hal-0.2.7/src/adc.rs:89:35
   |
89 | pub trait OneShot<ADC, Word, Pin: Channel<ADC>> {
   |                                   ^^^^^^^^^^^^ required by this bound in `stm32f3xx_hal::prelude::_embedded_hal_adc_OneShot::read`

error[E0277]: the trait bound `stm32f3xx_hal::gpio::Pin<Gpiob, U<15>, Analog>: stm32f3xx_hal::embedded_hal::adc::Channel<stm32f3xx_hal::pac::ADC4>` is not satisfied
  --> roadkill/src/main.rs:71:36
   |
71 |         let pressure_before: u16 = adc4.read(&mut pressure_before_pin).unwrap();
   |                                    ^^^^ ---- required by a bound introduced by this call
   |                                    |
   |                                    the trait `stm32f3xx_hal::embedded_hal::adc::Channel<stm32f3xx_hal::pac::ADC4>` is not implemented for `stm32f3xx_hal::gpio::Pin<Gpiob, U<15>, Analog>`
   |
   = help: the following other types implement trait `stm32f3xx_hal::embedded_hal::adc::Channel<ADC>`:
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<0>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC1>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<1>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC1>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<3>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC1>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<4>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<5>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<6>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpioa, U<7>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<ADC2>>
             <&stm32f3xx_hal::gpio::Pin<Gpiob, U<0>, Analog> as stm32f3xx_hal::embedded_hal::adc::Channel<stm32f3xx_hal::pac::ADC3>>
           and 34 others
   = note: required for `Adc<stm32f3xx_hal::pac::ADC4, stm32f3xx_hal::adc::OneShot>` to implement `stm32f3xx_hal::prelude::_embedded_hal_adc_OneShot<stm32f3xx_hal::pac::ADC4, _, stm32f3xx_hal::gpio::Pin<Gpiob, U<15>, Analog>>`

For more information about this error, try `rustc --explain E0277`.
warning: `roadkill` (bin "roadkill") generated 1 warning
error: could not compile `roadkill` due to 2 previous errors; 1 warning emitted
```

</details>

The datasheet indicates that PB15 is connected to ADC4 Channel 5.

[image]

Sh3Rm4n, 2023-02-19:
Thanks for reporting! :)  

Hm. This does indeed seem like a missing trait implementation, so that this port is supported.
I've originally generated those implementations from the bases of the stm32cubemx database. 
Apparently my script for that wasn't as fleshed out as I hoped to be, so I missed some channels.

See #337

closed 2023-02-19 by Sh3Rm4n

Sh3Rm4n, 2023-02-20:
I've released version v0.9.2, which includes the fixes. Can you report, if this fixes your problem? 

https://crates.io/crates/stm32f3xx-hal/0.9.2

PR #337 (2023-02-19): Add missing adc channels
Fixes #332