stm32-rs/stm32f3xx-hal #358: serial_dma.rs example broken in v0.10

tomhampshire, 2023-12-18:
`Serial` no longer has a `split()` function, as of version 0.10.
The [current version](https://github.com/stm32-rs/stm32f3xx-hal/blob/b0cead18208099b94a085fb634d0db4e75ac6e4d/examples/serial_dma.rs#L44) of the `serial_dma.rs` example is therefore broken.
Please, could someone advise on an alternative solution to using serial with DMA?
Many thanks.

Sh3Rm4n, 2024-01-06:
Yeah, this is indeed currently broken, sorry :( This was an oversight on releasing `v0.10.0`. 

I have an attempt for the fix at #359

tomhampshire, 2024-01-17:
Great - thank you!

robamu, 2024-03-23:
Does the new version allow to independently start RX and TX transfers? This is a use-case i have in my application where I am handling the reading of variable sized frames via DMA and the sending of frames in separate RTIC tasks.